basler/pypylon #765: Cannot set Gamma value for camera (AccessException)

SilverStarCoder, 2024-07-15:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am able to apply camera settings such as Exposure time, Gain, Pixel Format, etc.., but for some reason, I cannot set the gamma value, as it gives: `Node is not writable. : AccessException thrown in node 'Gamma' while calling 'Gamma.SetValue()' (file 'FloatT.h', line 85)`. I am of course ensuring that the camera is connected and open before applying settings. I thought it might be related to pixel format, but when using the emulated cameras, they don't give this issue on the same pixel format, (e.g. BayerRG8). What could be the possible causes? I have checked in the documentation and confirmed that the camera (_acA1920-40gc_) does indeed have a gamma feature node. 

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

acA1920-40g GigE camera, connected to Linux device running custom software. 

### Runtime information:

```shell
python: 3.8.10 (default, Nov 22 2023, 10:22:35)
[GCC 9.4.0]
platform: linux/aarch64/5.10.104-tegra
pypylon: 3.0.1 / 7.4.0.38864
```

thiesmoeller, 2024-07-16:
On your model you have to first enable the gamma feature..

https://docs.baslerweb.com/gamma?filter=Camera:acA1920-40gm#python_1

SilverStarCoder, 2024-07-16:
@thiesmoeller , it seems I might have incorrectly assumed that I didn't need to set the gamma selector value to "User" (I previously thought this was already the default value). After changing that, I am now able to set the gamma. Thank you.

closed 2024-07-16 by SilverStarCoder