basler/pypylon #766: Asynchronous grabbing from a 2-camera array

fhaibach, 2024-07-19:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

grabmultiplecameras.py is an example of synchronous grabbing of frames. I need to do asynchronous grabbing from two cameras. For example, lets label the cameras as "1" and "2", then the sequence might be
1 1 1 1 2 1 1 1 2 1 1 1 1 1
The pattern that interleaves 2 is irregular.
While I could get both images synchronously and just ignore unused images from 2, I was hoping for something cleaner and with less overhead.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

RPi 4, Raspbian/Debian OS
USB4 - direct connection via 30 cm cables

### Runtime information:

```shell
>>> import sys, pypylon.pylon, platform
>>> print(f'python: {sys.version}')
python: 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0]
>>> print(f'platform: {sys.platform}/{platform.machine()}/{platform.release()}')
platform: linux/aarch64/6.6.31+rpt-rpi-v8
>>> print(f'pypylon: {pypylon.pylon.__version__} / {".".join([str(i) for i in pypylon.pylon.GetPylonVersion()])}')
pypylon: 4.0.0 / 8.0.0.10
```

thiesmoeller, 2024-07-19:
Are the cameras HW triggered ?

fhaibach, 2024-07-22:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>

The camera can be software triggered. There is no hw triggering in this setup.

FgH...

Frederick Haibach
Sr. Director, Spectroscopy and Applications
Direct: 774.462.6711
***@***.***<https://www.kpmanalytics.com/>

SMA2016a, 2024-07-28:
are you using same camera model? and the resulting frame rate of both camera is same?

may be the camera 2 running slower or it is losing images.

HighImp, 2024-08-03:
Hi, you want the irregular pattern, but you dont know how to trigger the second camera irregular, did i get that correct? Can you tell the fps and when do you know if the second camera should take an Image? Is the irregular pattern known from the beginning or do you react to something during the process?

HighImp, 2024-08-03:
Also, you don't necessarily have to use the InstantCameraArray. 

You can also create two instances of InstantCameras and call camera.RetrieveResult for each camera individually.
If you use the Grab method: LatestImageOnly, pylon will automatically discard any uncaptured images. 

The cameras will continue to grab images in freerun. So if you want to avoid traffic to your host, a software trigger solution might be the best, but then you need to know the pattern in advance.  

One additional thing, because you say "grabmultiplecameras.py is an example of synchronous grabbing": 
The fact that both cameras start recording at the same time does not mean that they are necessarily synchronised. 
Over time, they will definitely be out of sync. 
To achieve a stable synchronised exposure, you need to use a trigger.
I understand that you want asynchronous images, but I want to be sure that you don't expect the exposure of the two cameras to occur simultaneously when an image is taken from both cameras.