basler/pypylon #767: Need high FPS but using .Array drastically slows down acquisition of data

neurorishika, 2024-07-21:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

# Problem Summary
I am trying to acquire data at 100 fps (less than max 106fps for camera) using pypylon however when I try to get the data and add it to a preallocated numpy array the acquisition loop slows down to almost half (using upcoming image strategy).

## Minimal Example (Upcoming Image)
```
import sys
import time
import numpy as np
from pypylon import pylon

N_GRAB = 1000
EXPOSURE_TIME, GAIN, CAMERA_FORMAT, WIDTH, HEIGHT, OFFSETX, OFFSETY = 10000, 30, "Mono8", 2048, 2048, 0, 0

data = np.zeros((N_GRAB, HEIGHT, WIDTH), dtype=np.uint8)
timestamps_a = np.zeros(N_GRAB, dtype=np.float64)
timestamps_b = np.zeros(N_GRAB, dtype=np.float64)

cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
print(f"Camera {cam.GetDeviceInfo().GetModelName()}_{cam.GetDeviceInfo().GetSerialNumber()} selected.")

assert not cam.IsUsb(), "Camera is USB.  Use GigE or CoaXPress cameras."

# Register the standard event handler for configuring software triggered single frame acquisition.
cam.RegisterConfiguration(pylon.AcquireContinuousConfiguration(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)

# Set MaxNumBuffer to 15
cam.MaxNumBuffer = 15

# Open the camera.
cam.Open()

# Turn off auto exposure and set exposure time
cam.ExposureAuto.SetValue("Off")
cam.ExposureTime.SetValue(EXPOSURE_TIME)

# Turn off auto gain and set gain
cam.GainAuto.SetValue("Off")
cam.Gain.SetValue(GAIN)

# get viable pixel formats
pixel_formats = cam.PixelFormat.Symbolics
assert CAMERA_FORMAT in pixel_formats, f"Invalid pixel format.  Must be one of {pixel_formats}"
cam.PixelFormat.SetValue(CAMERA_FORMAT)

# set width and height
cam.Width.SetValue(WIDTH)
cam.Height.SetValue(HEIGHT)
cam.OffsetX.SetValue(OFFSETX)
cam.OffsetY.SetValue(OFFSETY)

# get frequency
TSFREQ = cam.GevTimestampTickFrequency.GetValue()

# START OF VERSION 1 (NO ARRAY ALLOCATION)

# Start the grabbing  (no array is allocated)
cam.StartGrabbing(pylon.GrabStrategy_UpcomingImage)

for i in range(N_GRAB):
    # Queues a buffer for grabbing and waits for the grab to finish.
    grabResult = cam.RetrieveResult(1000, pylon.TimeoutHandling_ThrowException)
    timestamps_a[i] = grabResult.TimeStamp/TSFREQ
    # release the buffer
    grabResult.Release()

    # Check no grab result is waiting, because no buffers are queued for grabbing.
    if cam.GetGrabResultWaitObject().Wait(0):
        raise Exception("A grab result is waiting when none should be.")
[... 151 lines cut ...]
Frame rate: 37.96 +/- 1095.86 Hz (Max: 44.84 Hz, Min: 34.36 Hz)
```
I work with animals and the behavior I study is extremely fast so I need to acquire at high frame rates ideally with a software trigger to create a closed loop processing pipeline (Loop over [Trigger->Acquire->Process(Neural Network Prediction)->Manipulate Actuators] indefinitely) How do I fix this issue to acquire at the highest rate allowed by the pypylon?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

PC:
        CPU architecture:x86_64
        Operating System: Ubuntu Desktop 22.04 LTS 
        RAM: 128 GB

Interfaces used to connect the cameras:
        Camera: Basler ace2 a2A2448-105g5mBAS Monochrome 5GigE Camera with 3ft Cat 8 Ethernet Cable with M8 6pin Power Supply

### Runtime information:

```shell
python: 3.10.12 (main, Mar 22 2024, 16:50:05) [GCC 11.4.0]
platform: linux/x86_64/6.5.0-44-generic
pypylon: 4.0.0 / 8.0.0.10
```

SMA2016a, 2024-07-22:
firstly I wanted to point out that  "UpcomingImage" isn't correct gran strategy for your need.  

you dont need to pre allocate any buffers your self.

simply increase number of pylon buffer.

cam.MaxNumBuffer = N_GRAB

neurorishika, 2024-07-22:
Hi, could you explain why Upcoming Image is not a good idea?

SMA2016a, 2024-07-22:
The sdk will skip the images which are not retrieved in time. This causes reduced fps.

neurorishika, 2024-07-22:
I want to get just get the **latest image** from the camera. What should be the acqusition strategy?

All of them gave the same result:

Upcoming Image:
97.62764814995607 for Version 1 (no grabResult.Array)
97.62764814995607 for Version 2 (grabResult.Array used)

Latest Image Only:
97.62764814995607 for Version 1 (no grabResult.Array)
97.62764814995607 for Version 2 (grabResult.Array used)

I did these by creating new cameras for each version and calling it at the end:
 ```
# estimate the frame rate
print(f"{N_GRAB} frames acquired in {timestamps_a[-1]-timestamps_a[0]} seconds")
print(f"Frame rate: {1/np.diff(timestamps_a).mean():.2f} +/- {1/np.diff(timestamps_a).std():.2f} Hz (Max: {1/np.diff(timestamps_a).min():.2f} Hz, Min: {1/np.diff(timestamps_a).max():.2f} Hz)")

# print resulting frame rate
print(cam.ResultingFrameRate.GetValue())
```

I also tried estimating the effective frame rate and time taken just to **run grabResult.array** within the loop:
**Upcoming Image:**
[image]
[image]
Overload time being just the time taken to run grabResult.array

```
# Start the grabbing  (array is allocated)
cam.StartGrabbing(pylon.GrabStrategy_UpcomingImage)

overload = []

for i in range(N_GRAB):
    # Queues a buffer for grabbing and waits for the grab to finish.
    grabResult = cam.RetrieveResult(1000, pylon.TimeoutHandling_ThrowException)
    pre = time.perf_counter()
    if grabResult.GrabSucceeded():
        data[i] = grabResult.GetArray() # this is the only difference
        timestamps_b[i] = grabResult.TimeStamp/TSFREQ
    else:
        print(f"Frame {i} failed to grab due to : {grabResult.GetErrorDescription()}")
    overload.append(time.perf_counter()-pre)
    
    # release the buffer
    grabResult.Release()

    # Check no grab result is waiting, because no buffers are queued for grabbing.
    if cam.GetGrabResultWaitObject().Wait(0):
        raise Exception("A grab result is waiting when none should be.")
    
# Stop the grabbing
cam.StopGrabbing()
```
**Latest Image Only:**
[image]
[image]

neurorishika, 2024-07-22:
Okay, I tried something. Continuous Acquisition Configuration but used Latest Image only Strategy along with a large buffer size as you suggested. Performance immediately gets better!

**With grabResult.Array:**
```
Camera a2A2448-105g5m_40490701 selected.
1000 frames acquired in 10.233 seconds
Frame rate: 97.63 +/- 354197422.68 Hz (Max: 97.63 Hz, Min: 97.63 Hz)
97.62764814995607
```

[image]
[image]

**Quick question about this: Here in this example i only have a 1000 frames but in my experiments I need atleast 1 hr of data. i.e. 3600000 frames of data. Would the MaxNumBuffer need to be this big? Or does the Latest Image Strategy clear up the buffer after a result is grabbed?**

SMA2016a, 2024-07-23:
100fps*60*60s  * 5MB = 1.8TB Data in Total.  Your PC has only 128GB RAM. So keeping all frame in RAM will not work.
so you need to write the buffer to the harddisc but not in the grabloop.

Keep the Maxbuffer count to 1000
change your line to 

data[i] = result  // keep the grabresult in your queue.

and write the buffer to the harddisc in a separate thread. You need to run at least 2 or 3 worker threads parallel 

set the grab strategy   to onebyone 

please use such M2 harddrive to get good writing performance  or use multiple hard drives

https://www.amazon.de/Fikwot-FN950-Internes-Graphene-K%C3%BChlaufkleber/dp/B0C9DHGBK9/ref=sr_1_2_sspa?crid=11L3B9RTKQWVL&dib=eyJ2IjoiMSJ9.8TVbqvYvb-epgdJGz_nqAov5jTaLLLMNK-JDGVJGyTDbr2gZ78CZqYxr5hm3HOQYh0GUPgMYrJSyCh9Bc0-T05H-d_u88fvx91aJxdHuvAcsW2pZqdBf2QrSUl-34zPLgFPOxwJ7p8dQNKmiql0rTpYwgQnQO0f4J38S7efSfraQxWhQxDYfDSTD7EFACeLOXOI0VeAAjtyedY8S5UV7awc5NRc1GCvdJXsvuv-Cy0M.S3tSv03YgR2xN-LLSvmPo9W0Vx60DMEZd9vwbVQHvq8&dib_tag=se&keywords=m2+ssd+2tb&qid=1721708543&sprefix=m2+%2Caps%2C94&sr=8-2-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1

closed 2024-07-31 by SMA2016a