basler/pypylon #769: Delay in acquisition

bysjsys, 2024-08-06:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

My project is used for industrial vision detection, and the trigger wait time needs to be increased to a larger value. But this leads to a longer time (about 0.1s) between the camera trigger and receiving the picture. It seems not significant but did affect the detection efficiency. I want to improve the trigger wait time while reducing the delay to maintain a fast acquisition speed. I tried setting TriggerDelay but it didn't seem to work

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Windows 10, X86_64, acA2440-75uc

### Runtime information:

```shell
python: 3.9
pypylon: 7.1.0
```

thiesmoeller, 2024-08-06:
Do you use a hardware trigger?
What do you mean by trigger wait time?

bysjsys, 2024-08-07:
Yes, we use a hardware trigger. Trigger wait time is originally set as 5000ms in RetrieveResult(). I means there is a delay time between the camera is triggered and the graph is acquired in software. We found when the trigger wait time is set as a larger value, such as 1000000ms, this delay time will be incresed. Even if this delay time is short (0.1s) but still can not meet our needs with high efficiency.

thiesmoeller, 2024-08-07:
The value you give for RetrieveResult is only the maximum time to wait for an image to arrive. It has no impact on the latency from trigger to image.

The time you perceive is the time the camera needs to expose readout and transfer the image data.

What is your camera model, ROI, Pixelformat and exposure time ?

SMA2016a, 2024-08-08:
I think using Onimage Callback might be  also a good solution.