basler/pypylon #770: Intensity image and point cloud cannot be acquired

NTECHAIFA, 2024-08-07:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello.
What I want to do is to display a Intensity Image and Point Cloud as shown in the blaze viewer when running python.
 
I am using Balser blaze-101.
I ran the sample program but all I got was either the depth map being displayed or an error in the middle.
All that was displayed when the following was performed was depth data.
utilityimageformatconverter.py
save_image.py
opencv.py
guiimagewindow.py
 
Depth data was also displayed when the following was performed
basler/pypylon-samples/notebooks /deviceenumeration_and_configuration.ipynb

I can't make pylon viewer recognize the camera, but blaze viewer can display Intensity Image, Depth Map, and Point Cloud.

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup & camera model(s) used

Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz   3.60 GHz
RAM: 16.0 GB
OS : Windows 10

Camera: blaze-101
cable: GEV-C7M12-L10

### Runtime information:

```shell
python: 3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 4.0.0 / 8.0.0.10
```

SMA2016a, 2024-08-08:
This is correct that current pylon does not show the blaze camera. To display the images the opencv image windows is one of the good option

NTECHAIFA, 2024-08-09:
I have already succeeded in getting openCV to display images from the camera.
As per your first question, I was able to run some of the sample programs, including opencv.py. 
All of those that yielded images were depth data.
The intensity image could not be acquired.

Also, some lines in the sample program cause errors, for example, _genicam.LogicalErrorException when executing each of the following two.
If this error does not seem to be related to the matter of images and point clouds not being displayed, please ignore this error story.

camera.Gain.Value =42
print(camera.GainRaw)

  File "C:\...\site-packages\pypylon\pylon.py", line 4336, in __getattr__
    return self.GetNodeMap().GetNode(attribute)
  File "C:\...\site-packages\pypylon\genicam.py", line 1118, in GetNode
    return _genicam.INodeMap_GetNode(self, Name)
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16740)

Also, how can I get a point cloud on python as I could not find sample code to display a point cloud?

cpt-wojtech, 2024-08-19:
It seems like your are utilizing pypylon which does not support our blaze camera. Please install our current pylon version and the blaze supplementary package 

https://www2.baslerweb.com/de/downloads/downloads-software/software-pylon-7-5-0-windows/
https://www2.baslerweb.com/de/downloads/downloads-software/pylon-supplementary-package-for-blaze-1-6-0-windows/

Then you can find our python blaze samples in our sample folder, e.g. under the following path:

C:\Program Files\Basler\pylon 7\Development\Samples\blaze\Python

Our python blaze sample are using the Python harvester module. More information about the harvester module can be found here:

https://github.com/genicam/harvesters

With all this you should be able to retrieve and display the point cloud data. 

If you have any further questions later on, please contact your local Basler Support team.

https://www.baslerweb.com/en/support/contact/

cpt-wojtech, 2024-08-22:
@NTECHAIFA 
I just got the information that the newest pypylon release 4.0.0 actually does support the blaze. So it won't be necessary to use Harvesters. 
The attached samples demonstrate how to access and display depth and intensity data as well as the point cloud. Those pypylon samples will be included into the upcoming blaze supplementary package release. 

If you have further questions or need assistance, please feel free to contact your local Basler Support https://www.baslerweb.com/en/support/contact-overview/
[pypylon_blaze_samples.zip](https://github.com/user-attachments/files/16707977/pypylon_blaze_samples.zip)

NTECHAIFA, 2024-09-02:
I was able to successfully retrieve the point cloud data and image data. Thank you very much.

closed 2024-09-02 by NTECHAIFA