basler/pypylon #771: Basler PGI and Bayer pixel formats

loksbjerre, 2024-08-16:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Documentation states hat Basler PGI does not support the Bayer pixel formats. But when choosing a Bayer pixel format in Pylon Viewer, selecting PGI demosaicing is still possible, and it seems to work alright, i.e changing sharpening, noise reduction and scaling works ?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

a2A4096-30ucPRO, X86_64, Windows 11, 64 GB Ram, usb 3.2, 3 m usg cable

### Runtime information:

```shell
pylon info: 8.0.0.10 64-Bit
----------------------------------------
pylon Setup info: 7.5.0.15658 
----------------------------------------
pylon Application info: 3.3.0.15479 64-Bit
----------------------------------------
pylon Viewer info: 7.5.0.15479 64-Bit
----------------------------------------
Host system infos:

CPU architecture: x86_64
Kernel type: winnt
Kernel version: 10.0.22631
OS type: windows
OS version: 11
Render Backend: Direct3D11
----------------------------------------
Screen(s):

Screen 0 scale factor: 1
Screen 0 resolution: 3840x2160
Screen 1 scale factor: 1
Screen 1 resolution: 2560x1440
----------------------------------------
Window(s):

"pylon Viewer 64-Bit" on screen 0
----------------------------------------
Host network infos:

Host network adapter Ethernet:
    Name: ethernet_32769
    MAC address: B0:6E:BF:60:88:FC
    IP address: 192.168.2.122
    Subnet mask: 255.255.255.0
----------------------------------------
Device infos:

USB3 Vision device:
    Model name: a2A4096-30ucPRO
    Manufacturer name: Basler
    Serial number: 40525486
    Device GUID: 267601CB7EF5
    Product ID: 0xBA05
    Vendor ID: 0x2676
    USB version: 3.2.0
    Device path: \\?\usb#vid_2676&pid_ba05&mi_00#7&91a2638&0&0000#{79bfeeca-947f-11e1-bc58-001b2163ef96}
```

HighImp, 2024-08-16:
Hello, 
it is possible to select the demosaicing node PGI in a Bayer pixel format, but it is not recommended. If you use Auto mode, the best algorithm will be selected for you. But if you like the results with forced PGI on Bayer format, you can of course leave it enabled.  See also: https://docs.baslerweb.com/demosaicing-mode   
Best Regards

HighImp, 2024-08-16:
You will probably see severe artifacts that cannot be fixed due to this combination.

thiesmoeller, 2024-09-09:
PGI contains the steps of demosaicing _and_ denoising/sharpening.

If you enable PGI _and_ a Bayer output the camera will generate internally full RGB image and output this as a  Bayer raw image.

So you get denoising/sharpening and but the final demosacing and thus a relevant part of the image quality will depend on the software algorithm you use on your host.

basilbatov, 2025-02-07:
From [https://docs.baslerweb.com/demosaicing-mode#why-use-demosaicing-mode](https://docs.baslerweb.com/demosaicing-mode#why-use-demosaicing-mode):

_"Basler color cameras always perform demosaicing, even when the pixel format is set to a Bayer format. This is due to the camera's internal image processing."_

So how can I obtain this image that gets demosaiced internally by the camera? Selecting RGB pixel format significantly reduces the maximum framerate, but selecting the Bayer format and doing the demosaicing on the host increases the processing on the host, which is pointless if the demosaiced image is already produced by the camera anyway?