basler/pypylon #772: Weird GetTimeStamp() results

pourfard, 2024-08-24:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,

I get timestamp of the grab results using GetTimeStamp(). When the camera's FPS is 10, the difference between two consecutive frames is always 0.125 seconds. But i expect it should be 0.1 seconds. How can i get the actual timestamp of the frames in seconds or milliseconds.

`print(grabResult.GetTimeStamp() / 1e8)`

here is the output of the above code for some frames while i'm getting 10 frames per seconds:

```
921920.39391752
921920.51891786
921920.6439182
921920.76891854
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

PC:
* CPU: ARM64
* OS: Ubuntu
* RAM: 4GB

### Runtime information:

```shell
python: 3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0]
platform: linux/aarch64/5.10.110-rockchip-rk3588
pypylon: 4.0.0 / 8.0.0.10
```

thiesmoeller, 2024-08-24:
What is gives you the resulting frame period feature ?
What is your exposure time ?

There can be other factors reducing the framerate below your set frame rate 
Exposure time 
Transport layer configurations like throttled bandwidth

pourfard, 2024-08-24:
> What is gives you the resulting frame period feature ?
> What is your exposure time ?
> 
> There can be other factors reducing the framerate below your set frame rate 
> Exposure time 
> Transport layer configurations like throttled bandwidth 
> 

In fact, i am getting 10 frames  per second and there is no problem in fps, but grab result steps are larger.

Is it possible I get two similar frames with different ID? Each frame has unique ID.

Exposure is about 400.

How can I get period feature?

thiesmoeller, 2024-08-24:
As you didn't mention your camera model I assume, that you use a gige camera in standard ( not ieee1588 ) mode 

In this case the tick frequency of the timestamp is 125mhz not 1ghz

pourfard, 2024-08-24:
> As you didn't mention your camera model I assume, that you use a gige camera in standard ( not ieee1588 ) mode 
> 
> In this case the tick frequency of the timestamp is 125mhz not 1ghz

Yes, I have a gige camera. Can I convert it by multiplying to 8?

thiesmoeller, 2024-08-24:
You divide the timestamp you get from the çamera by 125e3 to get milli seconds

HighImp, 2024-08-26:
Hi, please use the value of "GevTimestampTickFrequency". It can change if you toggle extended network features, like PTP.

pourfard, 2024-08-27:
Thanks for your help.

closed 2024-08-27 by pourfard