basler/pypylon #776: Please write a comprehensive title

fkdlqj13245, 2024-09-03:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

    It has the error during the program run( it happened close with the ulimit open files umber).
    The device is not reachable or controlled by another application.

while True:
        idx += 1
        print('\nopen: ', idx)

        # open
        try:
            camera.Open()
        except Exception as e:
            print('[ERROR]: {0}'.format(traceback.format_exc()))
            break

        gc.collect()

        print('sys.getrefcount(instance):', sys.getrefcount(instance))
        print('sys.getrefcount(device):', sys.getrefcount(device))
        print('sys.getrefcount(camera):', sys.getrefcount(camera))

        # close
        camera.StopGrabbing()
        camera.Close()

        sleep(1)
#        sleep(10)

        print('closed')

    # close
    camera.StopGrabbing()
    camera.Close()

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

[PC Environment]

OS: Ubuntu 20.04(kernel version: 5.15.0-94-generic)

GPU: RTX3060
PYTHON: 3.8.10

### Runtime information:

```shell
pypylon version: 3.0.1 ( also tested 4.0.0)

pylon viewer: 7.4 64bit
```

SMA2016a, 2024-09-06:
This could be bug in GigE TL.   you can build pypylon against pylon 7.0  or I can build a wheel  if you let me your python version

fkdlqj13245, 2024-09-09:
Is the happened only pylon 7.4.0?
Here is the python version.
PYTHON: 3.8.10

SMA2016a, 2024-09-18:
[Is the happened only pylon 7.4.0?] yes. would you build your version against 7.0?

fkdlqj13245, 2024-10-23:
Thanks. It will test. I have one more question. Is the bug fixed in pylon 8 version?