basler/pypylon #777: BUG: AttributeError: SwigPyObject object has no attribute 'DataComponentCount'

mdauerap, 2024-09-07:
### Describe the issue:

I have a need to extract data from grab_results; I've noticed two issues:
1.  The grab_result `GetDataContainer` function returns a SwigPyObject that does not fully passthrough the DataContainer signature (see title or error message).
2. There is a memory leak as there is no destructor found.

Pylon: Tried both 7.2.0 with pypylon 2.3.0 and 7.5.0 with pypylon 3.0.1 (I am using py38) on a acA4024-8gc and a daA3840-45uc camera. Same results in all configurations.

This is preventing me from accessing the data container.

### Reproduce the code example:

```python
# Reference: https://github.com/basler/pypylon/blob/master/samples/grabdatacontainer.py

# ===============================================================================
#    This sample illustrates how to grab and process data using the CInstantCamera class.
#    The data is grabbed and processed asynchronously, i.e.,
#    while the application is processing a buffer, the acquisition of the next buffer is done
#    in parallel.
#
#    Utilizes the API for accessing GenICam Generic Data Container (GenDC).
#    This will allow the use of, e.g., Basler blaze 3D cameras.
# ===============================================================================
from pypylon import pylon
from pypylon import genicam

import sys

# Number of results to be grabbed.
countOfResultsToGrab = 100

# The exit code of the sample application.
exitCode = 0

try:
    # Create an instant camera object with the camera device found first.
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    camera.Open()

    # Print the model name of the camera.
    print("Using device ", camera.GetDeviceInfo().GetModelName())

    # demonstrate some feature access
    new_width = camera.Width.Value - camera.Width.Inc
    if new_width >= camera.Width.Min:
        camera.Width.Value = new_width

    # The parameter MaxNumBuffer can be used to control the count of buffers
    # allocated for grabbing. The default value of this parameter is 10.
    camera.MaxNumBuffer.Value = 5

    # Start the grabbing of c_countOfImagesToGrab grab results.
    # The camera device is parameterized with a default configuration which
    # sets up free-running continuous acquisition.
    camera.StartGrabbingMax(countOfResultsToGrab)

    # Camera.StopGrabbing() is called automatically by the RetrieveResult() method
    # when c_countOfImagesToGrab grab results have been retrieved.
    while camera.IsGrabbing():
        # Wait for grabbed data and then retrieve it. A timeout of 5000 ms is used.
        grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        # Data grabbed successfully?
        if grabResult.GrabSucceeded():
            # Get the grab result as a PylonDataContainer, e.g. when working with 3D cameras.
            pylonDataContainer = grabResult.GetDataContainer();
            print("Component Count: ", pylonDataContainer.DataComponentCount);
            # Access data components if the component type indicates image data
            for componentIndex in range(pylonDataContainer.DataComponentCount):
                pylonDataComponent = pylonDataContainer.GetDataComponent(componentIndex);
                if pylonDataComponent.ComponentType == pylon.ComponentType_Intensity:
                    # Access the component data.
                    print("PixelType: ", pylonDataComponent.PixelType)
                    print("SizeX: ", pylonDataComponent.Width)
                    print("SizeY: ", pylonDataComponent.Height)
                    print("OffsetX: ", pylonDataComponent.OffsetX)
                    print("OffsetY: ", pylonDataComponent.OffsetY)
                    print("PaddingX: ", pylonDataComponent.PaddingX)
                    print("DataSize: ", pylonDataComponent.DataSize)
                    print("TimeStamp: ", pylonDataComponent.TimeStamp)
                    img = pylonDataComponent.Array
                    print("Gray value of first pixel: ", img[0, 0])
                pylonDataComponent.Release()
            pylonDataContainer.Release()
        else:
            print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
        grabResult.Release()
    camera.Close()

except genicam.GenericException as e:
    # Error handling.
    print("An exception occurred.")
    print(e)
    exitCode = 1

sys.exit(exitCode)
```

### Error message:

```shell
Using device  acA4024-8gc
Traceback (most recent call last):
  File "basler_example.py", line 66, in <module>
    print("Component Count: ", pylonDataContainer.DataComponentCount);
AttributeError: 'SwigPyObject' object has no attribute 'DataComponentCount'
swig/python detected a memory leak of type 'Pylon::CPylonDataContainer *', no destructor found.
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

- OS
-- ARM64
-- Ubuntu 20.04 LTS
-- 8GB
- Cam
-- acA4024-8gc (GigE) - via PoE switch that is known to be working - Cat 5e
-- daA3840-45uc (USB) - builtin

### Runtime information:

```shell
python: 3.8.10 (default, Jul 29 2024, 17:02:10)
[GCC 9.4.0]
platform: linux/aarch64/5.10.104-rt63-tegra
pypylon: 3.0.1 / 7.4.0.38864
```

thiesmoeller, 2024-09-09:
We enabled the full/working support for datacontainers on **pypylon 4.0**

In your use-case, as you only have 2D cameras these are options:

* compile pypylon 4.0 for your target and python 3.8
   
   clone the repository and `pip install . `

* use the 2D standard way to access the grab result as shown in https://github.com/basler/pypylon/blob/master/samples/grab.py

mdauerap, 2024-09-09:
Thanks I'll give this a shot, closing ticket as it has been resolved in 4.0

closed 2024-09-09 by mdauerap