basler/pypylon #779: Grabbing data from blaze-101, receiving range data twice(?)

jens-siebert, 2024-09-12:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I'm running a system with a mixed camera setup (ace and blaze) in trigger mode. The software to control the cameras and for acquiring data from them is written in Python using pypylon 4.0.0.

For the ace cameras, I'm registering an ImageEventHandler which receives the image data from the cameras, converts them and pushes to an image processing pipeline for further processing.

For the newly added blaze cameras. I wanted to implement the same pattern. This works quite okay so far except for some problems I have when extracting the range, intensity and confidence data from the data container received as the grab result in the ImageEventHandler.

The code for this extraction follows the examples quite closely:
```python
for index in range(data_container.DataComponentCount):
   data_component = data_container.GetDataComponent(index)

  if not data_component.IsValid():
      logging.error(f"Data component {index} is invalid!")
      continue

  if data_component.ComponentType == pylon.ComponentType_Range:
      range_data = data_component.Array
      logging.debug(f"Received data component of type {data_component.ComponentType} and shape {range_data.shape}")
  elif data_component.ComponentType == pylon.ComponentType_Intensity:
      intensity = data_component.Array
      logging.debug(f"Received data component of type {data_component.ComponentType} and shape {intensity.shape}")
  elif data_component.ComponentType == pylon.ComponentType_Confidence:
      disparity = data_component.Array
      logging.debug(f"Received data component of type {data_component.ComponentType} and shape {disparity.shape}")

  data_component.Release()
```

However, the debug output looks as follows:
```shell
2024-09-12 17:36:34,864 Received data component of type 4 and shape (480, 640, 3)
2024-09-12 17:36:34,865 Received data component of type 4 and shape (480, 640)
2024-09-12 17:36:34,865 Received data component of type 6 and shape (480, 640)
```

As you can see, the component type 4 (range) has been received twice with different shapes.

I already tried to change several configuration parameters to see if this has any kind of effect but there was no change in this behavior.

Is this a bug or am I doing something wrong here?

By the way, the code to configure the camera is as follows:
```python
blaze_cam.TriggerMode.Value = "On"
blaze_cam.TriggerSource.Value = "Line0"
blaze_cam.TriggerSelector.Value = "FrameStart"

blaze_cam.ComponentSelector.Value = "Intensity"
blaze_cam.ComponentEnable.Value = True
blaze_cam.PixelFormat.Value = "Mono16"

blaze_cam.ComponentSelector.Value = "Confidence"
blaze_cam.ComponentEnable.Value = True
blaze_cam.PixelFormat.Value = "Confidence16"

blaze_cam.ComponentSelector.Value = "Range"
blaze_cam.ComponentEnable.Value = True
blaze_cam.PixelFormat.Value = "Coord3D_ABC32f"
```

Any help would be highly appreciated! Thanks!

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

PC or embedded system model/type:
- x64_64
- Ubuntu 22.04
- 64 GB RAM

Interfaces used to connect the cameras:
- internal network interfaces of the PC (Intel?), 1 GB/s

### Runtime information:

```shell
python: 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0]
platform: linux/x86_64/6.8.0-40-generic
pypylon: 4.0.0 / 8.0.0.65432
```

SMA2016a, 2024-09-13:
would you share the version number of the pylon Supplementary Package for blaze with us? so I can try same setup on my side

jens-siebert, 2024-09-13:
The problem has been solved already. I forgot to add the following line to my camera configuration code:

```python 
blaze_cam.GenDCStreamingMode.Value = "On"
```

After adding this line, everything works fine 👍

closed 2024-09-13 by jens-siebert