basler/pypylon #780: Smaller timestamp for newer frame

pourfard, 2024-09-18:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Is it possible to get a frame with smaller timestamp than previous frame? I have a GigE 2 Megapixel camera.

```python
        self.cam.AcquisitionMode.SetValue('Continuous')
        self.cam.Close()
        self.cam.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
```

I use above code to start grabbing and here is my code to grab a frame:

```python
def get_frame(self):
        for i in range(5):
            with self.cam.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:

                image = None
                grab_time = None

                if grabResult.GrabSucceeded():
                    print(grabResult.GetTimeStamp())
...
```
Sometimes frames not grabbed sequentially and the timestamp of newer frame is smaller than previous frame. Is it possible? If yes how can I solve it? I need ordered frames.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

PC:

CPU: ARM64
OS: Ubuntu
RAM: 4GB

### Runtime information:

```shell
python: 3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0]
platform: linux/aarch64/5.10.110-rockchip-rk3588
pypylon: 4.0.0 / 8.0.0.10
```

thiesmoeller, 2024-09-18:
This would only be possible if 

* You are synced to PTP in arbitrary time domain ( not UTC ) and you lost the connection to your grand master clock and the camera found another clock with a smaller time

* Your code resets the GigEVision time stamp ( active commands )

Please log the blockid too to debug this

pourfard, 2024-09-19:
Thanks for your response,

I have not set any NTP server for the camera. But the host system was syncing with GPS and another NTP server and I disabled  GPS now. May it cause the problem? Also I don't reset timestamp using commands.

I sometimes see this error too:

`Buffer Incompletely Grabbed: Error 3774873620 GX Status 0xe1000014`

Is it related to this problem?

SMA2016a, 2024-09-20:
No

SMA2016a, 2024-12-26:
This topic was closed 60 days after the last reply. New replies are no longer wished on this thread.

closed 2024-12-26 by SMA2016a