basler/pypylon #782: C++ Access violation writing locaiton

wolffman122, 2024-09-30:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

```
auto& factory = Pylon::CTlFactory::GetInstance();
m_Camera = std::make_unique<Pylon::CBaslerUniversalInstantCamera>(factory.CreateFirstDevice());
m_Camera->Open();
```

When I hit this code on the second line, I am getting numerous Access violation writing locations. 
```
MyApplication.exe has triggered a breakpoint.

Exception thrown at 0x12858E35 (PylonUtilityPcl_v7_4.dll) in MyApplication.exe: 0xC0000005: Access violation writing location 0x00000004.
Exception thrown at 0x12858E82 (PylonUtilityPcl_v7_4.dll) in MyApplication.exe: 0xC000001D: Illegal Instruction.
Exception thrown at 0x12859011 (PylonUtilityPcl_v7_4.dll) in MyApplication.exe: 0xC000001D: Illegal Instruction.
MyApplication.exe has triggered a breakpoint.

Exception thrown at 0x12859058 (PylonUtilityPcl_v7_4.dll) in MyApplication.exe: 0xC0000005: Access violation writing location 0x00000005.
Exception thrown at 0x128593DF (PylonUtilityPcl_v7_4.dll) in MyApplication.exe: 0xC0000096: Privileged instruction.
Exception thrown at 0x12859579 (PylonUtilityPcl_v7_4.dll) in MyApplication.exe: 0xC0000005: Access violation writing location 0x00000006.
Exception thrown at 0x12859614 (PylonUtilityPcl_v7_4.dll) in MyApplication.exe: 0xC0000096: Privileged instruction.
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\PylonGtc_v7_4_TL.dll'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\PylonGtc_v7_4_TL.dll'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\PylonGtc_v7_4_TL.dll'. 
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\ProducerGEV.cti'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\ProducerGEV.cti'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\ProducerGEV.cti'. 
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\gxapi_v14.dll'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\gxapi_v14.dll'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\gxapi_v14.dll'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\gxapi_v14.dll'
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\ProducerGEV.cti'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\ProducerU3V.cti'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\ProducerU3V.cti'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\ProducerU3V.cti'. 
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\uxapi_v13.dll'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\uxapi_v13.dll'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\uxapi_v13.dll'. 
'MyApplication.exe' (Win32): Loaded 'C:\Windows\SysWOW64\setupapi.dll'. 
'MyApplication.exe' (Win32): Loaded 'C:\Windows\SysWOW64\cfgmgr32.dll'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\setupapi.dll'
'MyApplication.exe' (Win32): Unloaded 'C:\Windows\SysWOW64\cfgmgr32.dll'
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\uxapi_v13.dll'
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\ProducerU3V.cti'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\PylonUsb_v7_4_TL.dll'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\PylonUsb_v7_4_TL.dll'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\PylonUsb_v7_4_TL.dll'. 
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\uxapi_v13.dll'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\uxapi_v13.dll'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\uxapi_v13.dll'. 
'MyApplication.exe' (Win32): Loaded 'C:\Windows\SysWOW64\setupapi.dll'. 
'MyApplication.exe' (Win32): Loaded 'C:\Windows\SysWOW64\cfgmgr32.dll'. 
'MyApplication.exe' (Win32): Loaded 'C:\Windows\SysWOW64\devobj.dll'. 
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\gxapi_v14.dll'. 
'MyApplication.exe' (Win32): Unloaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\gxapi_v14.dll'
'MyApplication.exe' (Win32): Loaded 'C:\Program Files\Basler\pylon 7\Runtime\Win32\gxapi_v14.dll'. 
```

 If I just choose in the debugger to run, the program will continue and the pylon calls all work.  I am able to get images form the Pylon software and while running my own application.  Any help would be appreciated.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

PC
11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz   2.50 GHz
16.0GB
Windows 10 Pro 22H2

NIC: Intel I219-LM

11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz   2.50 GHz

11th Gen Intel(R) Core(TM) i7-11850H @ 2.50GHz   2.50 GHz

### Runtime information:

```shell
python: 3.11.2 (tags/v3.11.2:878ead1, Feb  7 2023, 16:38:35) [MSC v.1934 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 4.0.0 / 8.0.0.10
```

thiesmoeller, 2024-09-30:
It seems that you mix 32bit and 64bit code.

Please check your environment that you use to compile.

wolffman122, 2024-09-30:
@thiesmoeller is there a way to set pylon to work with win32 or would i have to go back to a previous version of pylon that supports win32?

thiesmoeller, 2024-10-01:
pylon versions are 64bit only for quite some time

Latest version with 32bit support on Windows is 6.1.1