basler/pypylon #783: Need to restart the camera by resetting the PoE on the switch

stbnps, 2024-10-03:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

We have this issue where we stop being able to stream frames from the camera. Even though we sometimes can enumerate the camera in pylon viewer and pypylon, we cannot get frames using pylon viewer or pypylon, and this issue is only resolved after we reset PoE on the switch, so that the camera is restarted. Other times, we don't see the camera (and we don't even get ping replies even though it's connected), and we only see it back after we restart the PoE on the switch.

This issue only happens very sparingly, so it's really hard to reproduce. Do you think it's a bug in the camera firmware? A cabling issue? Maybe an issue with our switch? Do you recommend using the DeviceReset function when we can still enumerate the camera?

Thanks!

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup & camera model(s) used

CPU architecture: x86 64
Operating System: Ubuntu 22.04
RAM: 128 GB

Type in type/model/name of interface cards: NetXtreme-E BCM57504
Switches or hubs used: Netgear M4300
Cable types/lengths: Cat 6A

### Runtime information:

```shell
python: 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0]
platform: linux/x86_64/5.15.0-119-generic
pypylon: 3.0.1 / 7.4.0.38864
```

thiesmoeller, 2024-10-03:
As this should not happen at all.... finding the root cause is better than searching for fixing the symptoms.

Could you list the camera you use? ( Hardware setup & **camera model(s)** used)

stbnps, 2024-10-04:
The camera model is a2a1920-51gcpro. The camera is connected to the switch using a Cat 6A cable and, for some of them, we are using the Basler RJ45 with a screw connector. 

They are in an industrial environment and, in the past, we've seen some cables having small cuts because of the friction against sharp edges. Can this be related?

SMA2016a, 2024-12-26:
This topic was closed 60 days after the last reply. New replies are no longer wished on this thread.

closed 2024-12-26 by SMA2016a