basler/pypylon #784: digital overlap HDR on imx334 (daA3840-45uc)

dk-teknologisk-bgd, 2024-10-07:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

is there is functionality in the (py) Pylon to enable digital overlap HDR from the imx334 sony sensor? specifically on the dart daA3840-45uc board-level camera.
I am working on a project in which this would be of very high value, and since the image sensors supports the DOL hdr, I am hoping that pypylon also does (or would).

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

jetson orin NX
daA3840-45uc 

### Runtime information:

```shell
None
```

SMA2016a, 2024-10-10:
no.

JustinMBrown, 2024-11-24:
The industrial camera industry in general seems to be lagging behind when it comes to all thing computational photography. But what I don't get is if the hardware itself supports it, why is it seemingly so hard to "turn on"?

HighImp, 2024-11-25:
Hi @JustinMBrown, It is always a trade-off between power consumption / function / price for standard cameras. If we activate all functions (and support the calculation in the FPGA), we will need larger and/or more expensive hardware and the power consumption will increase. If a function is necessary, you can always contact our sales department to find a suitable solution: [baslerweb.com](https://www.baslerweb.com/en/capabilities/project-implementation/)  Best Regards