basler/pypylon #785: BUG: Pylon 8.0 doesn't work on Ubuntu 24

JustinMBrown, 2024-10-08:
### Describe the issue:

I have a fresh install of Ubuntu 24.04.1 LTS, and it won't open. I tried installing the mesa drivers but it says they don't exist.

### Reproduce the code example:

```python
It's not a code issue, but this is the only forum I can submit feedback.
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

CPU architecture: X86_64
OS: Ubuntu 24.04.1 LTS Desktop
RAM: 32GB

### Runtime information:

```shell
Ubuntu 24.04.1 LTS
```

bjoernrennfanz, 2024-10-09:
Hello please make sure you have installed the required packages according to the release notes

libgl1-mesa-dri, libgl1, libglx-mesa0, libxcb-xinerama0, and libxcb-xinput0

With best regards.

thiesmoeller, 2024-10-09:
Is it possible that you installed pylon deb file using dpkg ?

If you install via apt-get install ./pylon_xxx.deb the dependencies should be resolved automatically

SMA2016a, 2024-10-10:
you also need libxcb-cursor0  for pylon 8

nouvak, 2024-10-24:
I'm getting the same `Segmentation fault (core dumped)` crash when trying to execute Pylons 8.0 on Ubuntu 24.04.

The advice from the similar issue report (https://github.com/basler/pypylon/issues/760), if I execute the following command, pylonviewer starts fine:
```
QT_QPA_PLATFORM=xcb /opt/pylon/bin/pylonviewer
```

thiesmoeller, 2024-10-24:
Beside looking into it,.we have to add Wayland or X11 to query beside os version ...

nouvak, 2024-10-24:
Ubuntu version: 24.04.1 LTS
Version of mutter (Wayland display server): 46.0-1ubuntu9
X11 version:    11.0
X.Org version: 23.2.6

SMA2016a, 2024-11-23:
did you also install libxcb-cursor0?