basler/pypylon #786: No Devices Found in Pylon Viewer: Steps to Resolve the Issue

shimens, 2024-10-15:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I'm currently experiencing an issue where no devices are detected in the Pylon Viewer. What are the potential solutions to address this problem?

I recently acquired the a2a2840-67g5mUV camera and connected it via an Ethernet cable, but nothing appears on the Pylon Viewer or in the IP Configurator. Interestingly, I also have a SWIR camera from Basler, which is also an Ace2 series, and when I connect it with the same Ethernet cable, it works flawlessly.

Do these two cameras require different options for proper operation?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

SWIR : a2A2640 - 240gmSWIR

UV : a2A2840 - 67g5mUV

### Runtime information:

```shell
No devices found!
```

SMA2016a, 2024-10-15:
what is your operating system? if Linux, try to start pylon viewer as sudo.
would you post the output of lsusb command

SMA2016a, 2024-10-15:
then start pylon USB logging tool and post me the log

SMA2016a, 2024-10-15:
Sorry, you have a GigE Camera. Can you open Ip configuration tool and check if it sees the cameras. You may be disable your firewall.

shimens, 2024-10-15:
> Sorry, you have a GigE Camera. Can you open Ip configuration tool and check if it sees the cameras. You may be disable your firewall.

When connecting the SWIR camera, the image appears as expected. However, when connecting the UV camera, nothing shows up. Could this be a firewall issue?

[image]
[image]

shimens, 2024-10-15:
"이더넷" means "ethernet"

SMA2016a, 2024-10-15:
would you send me the serial number of the UV camera via private message,please

shimens, 2024-10-16:
> would you send me the serial number of the UV camera via private message,please
> 
> does the camera gets warm if you connect the camera to network? do do power the camera via POE?

No, the camera does not get warm when connected via an Ethernet cable, and the LED does not turn on.
The SWIR camera, however, operates normally, and all these issues are occurring only with the UV camera.

Could you provide an email or other contact method so I can send the UV camera serial number privately?

shimens, 2024-10-16:
I've just disabled the firewall and tried again, but it's still showing 'no devices found'

closed 2024-10-21 by shimens

venkatyash1523, 2025-04-29:
When connecting the SWIR camera, the image appears as expected. However, when connecting the UV camera, nothing shows up. Could this be a firewall issue?

venkatyash1523, 2025-04-29:
I'm currently experiencing an issue where no devices are detected in the Pylon Viewer. What are the potential solutions to address this problem?

venkatyash1523, 2025-04-29:
> did you contact your local Basler Support or distributor to get support on
> this issue?

Kindly send contact details

venkatyash1523, 2025-04-29:
3days line stop kindly solve

SMA2016a, 2025-04-29:
https://www.baslerweb.com/de-de/support/kontakt/

SMA2016a, 2025-05-01:
I do not find any entries or private message right now.