basler/pypylon #787: Re: No Devices Found in Pylon Viewer: Steps to Resolve the Issue

shimens, 2024-10-17:
> would you send me the serial number of the UV camera via private message,please
> 
> does the camera gets warm if you connect the camera to network? do do power the camera via POE?

No, the camera does not get warm when connected via an Ethernet cable, and the LED does not turn on.
The SWIR camera, however, operates normally, and all these issues are occurring only with the UV camera.

Could you provide an email or other contact method so I can send the UV camera serial number privately?

_Originally posted by @shimens in https://github.com/basler/pypylon/issues/786#issuecomment-2415391721_

SMA2016a, 2024-10-18:
let view your email address in your profile, so can send you a private message.

shimens, 2024-10-18:
> let view your email address in your profile, so can send you a private message

I've switched my email visibility to public. Could you please check again?

shimens, 2024-10-18:
If it isn't visible, could you kindly contact me at [elec7230@gmail.com]?

SMA2016a, 2024-10-18:
sure

SMA2016a, 2024-10-18:
next step in progerss

closed 2024-10-18 by SMA2016a

shimens, 2024-10-18:
I’m afraid I might be misunderstanding, but could you clarify what you
mean by 'next step in progress'? As far as I’m aware, nothing has been
resolved yet.