basler/pypylon #790: BUG: Arguments exception when convert numpy to pylon image

wind0952, 2024-10-29:
### Describe the issue:

The user want to use vTools and OpenCV together. The workflow as follows.
* Pre-processing by OpenCV
* Convert numpy to pylon image
* Matching by vTools
It got arguments exception when convert numpy to pylon image by pylonimage.AttachUserBuffer.
But it seems worked in [Unable to Decompress Images in ROS2 using Pylon SDK in Python · Issue #778 · basler/pypylon (github.com](https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fbasler%2Fpypylon%2Fissues%2F778&data=05%7C02%7C%7C5b5cd58564c6466ad14908dcf7325a09%7C744babbe24234ffca968eb9943b74386%7C0%7C0%7C638657042872227716%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C0%7C%7C%7C&sdata=3HlgXXYaKRk6Sz1Q14UalhKt9KkEXcVDevU57mzsbAk%3D&reserved=0)
Could you check it, please?

### Reproduce the code example:

```python
import os
from pypylon import pylon
from pypylon import genicam
import cv2
import sys
# Simulate pre-processing
img = cv2.imread("barcode01.png",-1)

pylonImg = pylon.PylonImage()
# Got arguments exception
pylonImg.AttachUserBuffer(
    img.ctypes.data,
    img.shape[1]*img.shape[0],
    pylon.PixelType_Mono8,
    img.shape[1],
    img.shape[0],
    0,
    pylon.ImageOrientation_TopDown
)
```

### Error message:

```shell
Wrong number or type of arguments for overloaded function 'PylonImage_AttachUserBuffer'.
  Possible C/C++ prototypes are:
    Pylon::CPylonImage::AttachUserBuffer(void *,size_t,Pylon::EPixelType,uint32_t,uint32_t,size_t,Pylon::EImageOrientation,CPylonImageUserBufferEventHandler *)
    Pylon::CPylonImage::AttachUserBuffer(void *,size_t,Pylon::EPixelType,uint32_t,uint32_t,size_t,Pylon::EImageOrientation)
    Pylon::CPylonImage::AttachUserBuffer(void *,size_t,Pylon::EPixelType,uint32_t,uint32_t,size_t)
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

* CPU architecture: X86_64
* Operating System: Windows 10/11
* RAM: 32 GB

### Runtime information:

```shell
windows 11, pylon 8.0
python: 3.12.7 | packaged by Anaconda, Inc. | (main, Oct  4 2024, 13:17:27) [MSC v.1929 64 bit (AMD64)]
platform: win32/AMD64/11
pypylon: 4.0.0 / 8.0.0.10

windows 10, pylon 7.5
python: 3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)]
platform: win32/AMD64/10 
pypylon: 4.0.0 / 8.0.0.10
```

SMA2016a, 2024-10-29:
We need to workaround this issue. I will send you a solution directly

thiesmoeller, 2024-10-30:
we added the missing functionality currently in the branch [feature/fix_attachuserbuffer](https://github.com/basler/pypylon/tree/feature/fix_attachuserbuffer)

Sample code is here:
https://github.com/basler/pypylon/blob/d738c176d7d46b7587997c34dd5a1e2a4b91e3d9/tests/pylon_tests/emulated/pylonimage_test.py#L81

wind0952, 2024-11-01:
It worked. Thank you!

closed 2024-11-01 by wind0952