basler/pypylon #791: How to improve capture delay

Ting-Jia-Hu, 2024-11-04:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,
 
Sorry for my poor English.

I have two cameras “a2A1920-160ucBAS” to implement real-time target detect, now I implement it with "NVIDIA Jetson Orin NX 16G" and the time of processing the two pictures at the same time is at most 20ms, But there is a serious problem, it have a long delay.

My experiment setting is use two cameras to capture image and then processing these two images with OpenCV in C++, then calculate the target's coordinate and transmit data to my MPU (STM32F407) with CAN bus. I compare this data with ground truth, and it delays about 500ms like phase lag. It means every image I get is 500ms ago. But I only take 20ms to do image processing and CAN bus transmit. Is the other time spend on image transmit？Following picture is my experiment result. 
[image]

Actually, no matter I use Pylon Viewer or Pylon API in C++ , when I display these two images, it has a visible delay. My camera can reach highest 160 fps, and I connect it with "NVIDIA Jetson Orin NX 16G" then open it  with Pylon Viewer, it has about 156fps and not obvious delay. But when I connect two cameras both decrease to 93fps, I can understand it is because the embedded system's bandwidth limit, but why it have a long and visible delay？Because I need to slap the target, so real-time stereo vision is necessary.

Please assist me in solving this problem.
Thank you and sorry for my poor English again.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Embedded system : NVIDIA Jetson Orin NX 16G
Interface : USB cable

### Runtime information:

```shell
pylon Viewer 7.4.0.14100
```

Ting-Jia-Hu, 2024-11-07:
Hi,

Thank you very much for assisting me in the problem. The effect has been significantly improved.

thiesmoeller, 2024-11-08:
Please close the issue, if it is resolved for you.

closed 2024-11-09 by Ting-Jia-Hu