basler/pypylon #797: BUG: Segmentation fault calling CreateFirstDevice()

Niv-R, 2024-12-02:
### Describe the issue:

After upgrading to pypylon==4.1.0 from 4.0.0 using pip, I consistently encounter a "Segmentation fault" every time I call CreateFirstDevice() within a Docker container.

Downgrading to pypylon==4.0.0 resolves the issue, and the segmentation fault does not occur in this version.

This issue is specific to Docker, as it does not reproduce on the host OS. Both the Docker environment and the host OS are running Ubuntu 22.04 with python 3.11.7, with Docker configured to run in privileged mode.

### Reproduce the code example:

```python
from pypylon import pylon
tl = pylon.TlFactory.GetInstance()
tl.CreateFirstDevice()
```

### Error message:

```shell
Segmentation fault (core dumped)

0x00007fffe4ec3449 in arena_for_chunk (ptr=0x555555d6c870) at ./malloc/arena.c:156
156	./malloc/arena.c: No such file or directory.

gdb backtrace:

#0  0x00007fffe4ec3449 in arena_for_chunk (ptr=0x555555d6c870) at ./malloc/arena.c:156
#1  arena_for_chunk (ptr=0x555555d6c870) at ./malloc/arena.c:160
#2  __GI___libc_free (mem=<optimized out>) at ./malloc/malloc.c:3390
#3  0x00007fffc9814213 in ?? () from /usr/local/lib/python3.11/site-packages/pypylon/pylon-libusb-1.0.27.so
#4  0x00007fffc9814285 in ?? () from /usr/local/lib/python3.11/site-packages/pypylon/pylon-libusb-1.0.27.so
#5  0x00007fffc9814ba5 in ?? () from /usr/local/lib/python3.11/site-packages/pypylon/pylon-libusb-1.0.27.so
#6  0x00007fffc9807086 in libusb_init_context () from /usr/local/lib/python3.11/site-packages/pypylon/pylon-libusb-1.0.27.so
#7  0x00007fffc9c92abe in ?? () from /usr/local/lib/python3.11/site-packages/pypylon/libuxapi.so.14
#8  0x00007fffc9c745d9 in uxapi::Init() () from /usr/local/lib/python3.11/site-packages/pypylon/libuxapi.so.14
#9  0x00007fffca28139c in Create () from /usr/local/lib/python3.11/site-packages/pypylon/libpylon_TL_usb.so
#10 0x00007fffe3ba215f in ?? () from /usr/local/lib/python3.11/site-packages/pypylon/libpylonbase.so.9
#11 0x00007fffe3b9eac1 in Pylon::CTlFactory::CreateTl(Pylon::CTlInfo const&) () from /usr/local/lib/python3.11/site-packages/pypylon/libpylonbase.so.9
#12 0x00007fffe3b9f16e in Pylon::CTlFactory::EnumerateDevices(Pylon::DeviceInfoList&, Pylon::DeviceInfoList const&, bool) () from /usr/local/lib/python3.11/site-packages/pypylon/libpylonbase.so.9
#13 0x00007fffe3b9f8bc in Pylon::CTlFactory::InternalCreateDevice(Pylon::CDeviceInfo const&, GenICam_3_1_Basler_pylon::gcstring_vector const&, bool) ()
   from /usr/local/lib/python3.11/site-packages/pypylon/libpylonbase.so.9
#14 0x00007fffe3b9f6fd in Pylon::CTlFactory::CreateFirstDevice(Pylon::CDeviceInfo const&) () from /usr/local/lib/python3.11/site-packages/pypylon/libpylonbase.so.9
#15 0x00007fffe4866ea1 in _wrap_TlFactory_CreateFirstDevice () from /usr/local/lib/python3.11/site-packages/pypylon/_pylon.abi3.so
#16 0x00007fffe5260588 in cfunction_call (func=0x7fffe4acc450, args=<optimized out>, kwargs=<optimized out>) at Objects/methodobject.c:553
#17 0x00007fffe523a1dd in PyObject_Call () from /usr/local/lib/libpython3.11.so.1.0
#18 0x00007fffe529ecd0 in do_call_core (use_tracing=0, kwdict=0x0, callargs=0x7fffe4afeb90, func=0x7fffe4acc450, tstate=<optimized out>) at Python/ceval.c:7324
#19 _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=<optimized out>, throwflag=<optimized out>) at Python/ceval.c:5376
#20 0x00007fffe5295daf in _PyEval_EvalFrame (throwflag=0, frame=0x7ffff7fad020, tstate=0x7fffe55bbbd8 <_PyRuntime+166328>) at ./Include/internal/pycore_ceval.h:73
#21 _PyEval_Vector (tstate=0x7fffe55bbbd8 <_PyRuntime+166328>, func=<optimized out>, locals=0x7fffe4beb300, args=0x0, argcount=<optimized out>, kwnames=<optimized out>) at Python/ceval.c:6434
#22 0x00007fffe5317d1c in PyEval_EvalCode (co=co@entry=0x7fffe4bbc880, globals=globals@entry=0x7fffe4beb300, locals=locals@entry=0x7fffe4beb300) at Python/ceval.c:1148
#23 0x00007fffe532ded4 in run_eval_code_obj (tstate=0x7fffe55bbbd8 <_PyRuntime+166328>, co=0x7fffe4bbc880, globals=0x7fffe4beb300, locals=0x7fffe4beb300) at Python/pythonrun.c:1710
#24 0x00007fffe532de5b in run_mod (mod=<optimized out>, filename=<optimized out>, globals=0x7fffe4beb300, locals=0x7fffe4beb300, flags=<optimized out>, arena=<optimized out>) at Python/pythonrun.c:1731
#25 0x00007fffe532e531 in pyrun_file (fp=fp@entry=0x555555ebe000, filename=filename@entry=0x7fffe49f75a0, start=start@entry=257, globals=globals@entry=0x7fffe4beb300, locals=locals@entry=0x7fffe4beb300, 
    closeit=closeit@entry=1, flags=0x7fffffffdab8) at Python/pythonrun.c:1626
#26 0x00007fffe532e1c8 in _PyRun_SimpleFileObject (fp=0x555555ebe000, filename=0x7fffe49f75a0, closeit=1, flags=0x7fffffffdab8) at Python/pythonrun.c:440
#27 0x00007fffe532dff7 in _PyRun_AnyFileObject (fp=fp@entry=0x555555ebe000, filename=filename@entry=0x7fffe49f75a0, closeit=closeit@entry=1, flags=flags@entry=0x7fffffffdab8) at Python/pythonrun.c:79
#28 0x00007fffe533685c in pymain_run_file_obj (skip_source_first_line=<optimized out>, filename=0x7fffe49f75a0, program_name=0x7fffe4beb570) at Modules/main.c:360
#29 pymain_run_file (config=0x7fffe55a1c20 <_PyRuntime+59904>) at Modules/main.c:379
#30 pymain_run_python (exitcode=0x7fffffffdab0) at Modules/main.c:601
#31 Py_RunMain () at Modules/main.c:680
#32 0x00007fffe53363ed in Py_BytesMain (argc=<optimized out>, argv=<optimized out>) at Modules/main.c:734
#33 0x00007fffe4e47d90 in __libc_start_call_main (main=main@entry=0x555555555060 <main>, argc=argc@entry=2, argv=argv@entry=0x7fffffffdd08) at ../sysdeps/nptl/libc_start_call_main.h:58
#34 0x00007fffe4e47e40 in __libc_start_main_impl (main=0x555555555060 <main>, argc=2, argv=0x7fffffffdd08, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdcf8)
    at ../csu/libc-start.c:392
#35 0x0000555555555095 in _start ()
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Camera used: a2A2448-23gmBAS

PC: X86_64 Ubutu 22.04 - 32GB RAM
Interfaces: Ethernet, direct connection

### Runtime information:

```shell
The full Basler pylon Camera Software Suite is not installed inside the docker

python: 3.11.7 (main, Sep 11 2024, 13:52:27) [GCC 11.4.0]
platform: linux/x86_64/6.5.0-41-generic
pypylon: 4.1.0 / 9.0.3.215
```

thiesmoeller, 2024-12-02:
Can you show the full docker run cmdline ?

bjoernrennfanz, 2024-12-03:
Hi, can you try to install libusb package from Ubuntu (libusb-1.0) and remove the libusb from the pypylon package.

```shell
cd $(python3 -m site --user-site)/pypylon
rm -rf pylon-libusb-1.0.27.so
sudo apt-get install libusb-1.0
```

Niv-R, 2024-12-03:
Sorry for the delayed response; I needed additional time to perform thorough tests and pinpoint the source of the issue.
> Can you show the full docker run cmdline ?

For testing, I used the following command:
`docker run -it --network host --privileged <image> bash`

> Hi, can you try to install libusb package from Ubuntu (libusb-1.0) and remove the libusb from the pypylon package.
> 
> ```shell
> cd $(python3 -m site --user-site)/pypylon
> rm -rf pylon-libusb-1.0.27.so
> sudo apt-get install libusb-1.0
> ```
> 
> In this case, pylon will fallback to the libusb from the system.

I tried this approach, but unfortunately, it did not resolve the issue.

After thoroughly reviewing all the Docker image build steps, I identified the conflicting line that caused the issue with pypylon==4.1.0:
`export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libtcmalloc.so.4`

It seems that this optimization caused the segmentation fault with pypylon==4.1.0, even though pypylon==4.0.0 worked fine with the same configuration.

SMA2016a, 2024-12-04:
since the camera is a GigE one, it is easier to remove /usr/local/lib/python3.11/site-packages/pypylon/libpylon_TL_usb.so just for testing. Of course make a backup of the file