basler/pypylon #799: Save image using Pylon v tool

neerja123-gupta, 2024-12-05:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi, I am using the Pylon Viewer V tool to experiment with image processing techniques such as image enhancement and smoothing. I can create and save a recipe in the tool, but I am unable to save the resulting images. For example, if I want to save the camera image after applying the Image Enhancer, how can I do this using the V tool GUI?

[image]

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Windows

### Camera(s) used

Basler acA2040-35gm

### Runtime information:

```shell
Yes I have installed pylon viewer 64 bit release 8. And I was using Demo license.
```

SMA2016a, 2024-12-07:
<how can I do this using the V tool GUI?> Yes, you can do that. You can access the output image and save it. if you need more help, you can create need code line for c++ using the code generator in pylon viewer.  if you need more help, feel free to contact your local support team.

[image]

SMA2016a, 2024-12-09:
by the way, you need Eval  license to access the vTools via SDK, which is also free of charge for 180day. Send me private message or an email to your local support to get one.

neerja123-gupta, 2024-12-09:
Hi @SMA2016a , thank you for your response. I can save the recipe, but I’m still unable to save the image as a PNG, JPG, or any other image format, even with the output. Could this issue be related to using a demo license? Perhaps it only works with an evaluation license. 
[image]

bjoernrennfanz, 2024-12-09:
@neerja123-gupta You can save the output image of an vTools Recipe as PNG or other images formats using the C++ SDK or pypylon. In this case you nee a vTools Eval License, you can get one from your local Basler support.

neerja123-gupta, 2024-12-09:
@bjoernrennfanz ok, thank you!