basler/pypylon #800: Sending C++ commands to framegrabber using PyPylon

OptoIntel-mik, 2024-12-06:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I have a CXP-12 1C framegrabber that is connected to a Basler Boost camera and I am trying to write a program that acquires images from it using PyPylon. Sadly, it seems like some of the commands specific to interfacing with the framegrabber such as changing some of the transport layer parameters, as well as controlling the GPIO lines, are only available in C++ and do not have a PyPylon implementation. I was wondering if there is a method for me to use those C++ function calls from PyPylon or if I'm out of luck and need to write the full application in C++ in order to have access to those framegrabber specific functions. I know there might be a way to get around it by installing the full framegrabber SDK and using the python wrapper, but it seems like all the functions I need are already in the Pylon Software Suite API that is already installed, just not available in PyPylon. I thought I would check if there was a way to get around installing a whole new SDK which seems a bit tricky to get working on Linux because of kernel issues... Thanks!

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

CXP-12 1C Framegrabber
Basler Boost boA6500-36cm

### Camera(s) used

camera.DeviceFirmwareVersion.Value: p=xc2b_xgs32m/s=r/v=2.5.0/i=9015.39/h=43a1549

### Runtime information:

```shell
Using 
Pylon Release 8.0.0.16021
Pylon C++ SDK 9.0.3.215
Pylon Application 3.3.0.16064
Pylon Viewer 8.1.1.16064

Running on 
Ubuntu 24.04.1 LTS
```

thiesmoeller, 2024-12-06:
If you can access the feature from pylon sdk in c++ it is available in pypylon too.

Can you give an example of the access in c++ ?

OptoIntel-mik, 2024-12-06:
Yes, for example, I was trying to set the sensor height of the transport layer. I know it can be done directly from the Pylon Viewer, and the C++ command I am trying to send is
camera.GetTLParams().SensorHeight.SetValue(4948);

How would I go about doing this in pypylon?

OptoIntel-mik, 2024-12-06:
If that's of any help, the exact error I get when trying to change these parameters using pypylon is the following: 
LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16755)

SMA2016a, 2024-12-08:
Node not existing   -> your camera does not have the parameter

The easiest way is using pylonviewer check if the node is available then copy the needed code in line from the feature descreption. 

since the issue is nothing related to pypylon, feel free to conect your local support team to get better hardware support

thiesmoeller, 2024-12-08:
Tlparams is tlnodemap in pypylon

HighImp, 2024-12-08:
Hi, please try this:
```python
# ensure open camera object here
assert camera.IsOpen()

tl_nodemap = camera.GetTLNodeMap()
# change <node_name> with your desired feature
node = tl_nodemap.GetNode("<node_name>")

#  Manipulate the node value
print(node.Value)
node.Value = <value>
print(node.Value)
```

Best regards

OptoIntel-mik, 2024-12-09:
"The easiest way is using pylonviewer check if the node is available then copy the needed code in line from the feature description."
That's what I had done. The "camera.GetTLParams().SensorHeight.SetValue(4948);" that I posted was taken directly from the Pylon Viewer, and since I was able to modify it from Pylon Viewer I was confused as to why it was telling me that the node was not existing. 

I was able to get it to work using the code snippet provided by HighImp. Thanks! 
For reference if anybody else is encountering the same issue at some point, this is the code I am using to change the Height and SensorHeight parameters from the transport layer. 

```python
tl_nodemap = camera.GetTLNodeMap()
height_node = tl_nodemap.GetNode("Height")
sensorHeight_node = tl_nodemap.GetNode("SensorHeight")
height_node.Value = 4948
sensorHeight_node.Value = 4948