basler/pypylon #801: Can't save the UserDefinedName to the camera

alon-ba, 2024-12-19:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I try to set a custom name to the camera, using `SetUserDefinedName()`:

```py
from pypylon import pylon  # version 4.1.0

baslers = pylon.TlFactory.GetInstance().EnumerateDevices()
first_basler = baslers[0]
first_basler.GetUserDefinedName()       # returns an empty string
first_basler.SetUserDefinedName("My_Basler")
first_basler.GetUserDefinedName()       # returns "My_Basler"
```

But it has nothing to do with the real camera!
It only modifies the python object.
How do I commit this name to the camera?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

MacOS 14.6.1 (23G93)

MacBook Air M3

### Camera(s) used

you snippet doesn't work:
```
tl_factory = py.TlFactory.GetInstance() for dev in tl_factory.EnumerateDevices():
                                            ^^^
SyntaxError: invalid syntax
```

### Runtime information:

```shell
python: 3.10.8 (v3.10.8:aaaf517424, Oct 11 2022, 10:14:40) [Clang 13.0.0 (clang-1300.0.29.30)]
platform: darwin/arm64/23.6.0
pypylon: 4.1.0 / 7.3.2.1
```

SMA2016a, 2024-12-19:
you need to open the camera object and then set the userdefine name to the camera object.

alon-ba, 2024-12-19:
Could you please write a snippet, how would you do that?
Because i have tried it, and it didn't work for me.

SMA2016a, 2024-12-19:
sure, do you have usb or GigE camera?

alon-ba, 2024-12-22:
@SMA2016a 
It actually raises two questions for me - 
1. Is the reset mandatory? or just to make a point, to show that the property persists?
2. What is the different between DeviceUserId and UserDefinedName?

SMA2016a, 2024-12-22:
1) a USB camera requires a reset to take over the new setting but not the GigE. Since I do not know which camera you have, I reset the camera.
2) The GeniCam Node name is DeviceUserId. Cameras parameter names are defined according to GeniCam naming conversion. UserDefinedName is just used in device object in pylon only.

SMA2016a, 2024-12-22:
problem solved

closed 2024-12-22 by SMA2016a