basler/pypylon #804: Open raw files

Nemarott, 2025-01-06:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello, I'm working with raw files wich I analyze with Python. This works. My problem is that I want to show the pictures to other people but the standard program to open pictures on my Windows Pc can't open them. What program do I nee to open raw files or better is there some python code I can run to convert the raw files to a png for example, so I can show it to other people?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Windows

### Camera(s) used

Basler acA3088-57um

### Runtime information:

```shell
.
```

thiesmoeller, 2025-01-06:
From where do you save the raw file?
From PylonViewer?!  Or from a python script?

Nemarott, 2025-01-06:
> From where do you save the raw file? From PylonViewer?! Or from a python script?

Currently from the PylonViewer but in future from a python script.

HighImp, 2025-01-06:
Hi, 
you can use ImageJ: 
https://docs.baslerweb.com/knowledge/how-to-open-a-mono12-raw-image-with-imagej

Best Regards

HighImp, 2025-01-06:
Why don't you use TIFF instead of RAW? It's lossless and can be processed with OpenCV afterwards. Without color, no preprocessing is even required.

Nemarott, 2025-01-06:
> Why don't you use TIFF instead of RAW? It's lossless and can be processed with OpenCV afterwards. Without color, no preprocessing is even required.

I'm using the camera to see how my light from a light source is reflected from a Photo Multiplier Tube. For this I add up all the Gray Values of each pixel of the image. I made some example image and for some reason the sum of gray values with TIFF was like 10 times higher than the sum of gray values of the RAW file. Since the file size for TIFF and Raw was the same in my case I procceded with RAW.

HighImp, 2025-01-06:
Could it be that you're running into an issue with the alignment? 
As far as I know, TIFF uses the upper 12 bits for data, while RAW uses the lower ones. 
So, if your TIFF value is exactly 16 (2<<4) times higher than the RAW value, that could be the case. 

Personally, I would really recommend using TIFF to avoid the need to share code with everyone who wants to view your images. However, if you insist, the next response will include an example code. 
;-)
Best Regards!

Nemarott, 2025-01-07:
> Could it be that you're running into an issue with the alignment? As far as I know, TIFF uses the upper 12 bits for data, while RAW uses the lower ones. So, if your TIFF value is exactly 16 (2<<4) [...]

I just wanted to show the pictures to other pepople on my laptop, so the code would be only for me. But indeed the TIFF value is indeed 16 times higher than the RAW value I will provide the code I used to compare the values. It would be nice if you can  tell me what I need to change so this 16 times difference does not occur.

Thanks and Best Regards

`import numpy as np
from PIL import Image, ImageOps

#Tür zu Objekt 8bit 

#raw file einlesen
file_path="C:/Users/adria/Desktop/Testbilder für Pixel Format/neu/Raw Bild Expo 0,1.raw"
with open(file_path, 'rb') as file:
    # Dateiinhalt einlesen
    daten = file.read()
#daten in 2d Array umwandeln
height = 2064
width = 3088
array_raw = np.frombuffer(daten, dtype=np.uint16).reshape((height, width))

#andere Bilder laden und als array speichern

img_dc_m_8bit_bmp = Image.open('C:/Users/adria/Desktop/Testbilder für Pixel Format/neu/Bmp Bild Expo 0,1.bmp')
array_bmp=np.array(img_dc_m_8bit_bmp)

img_dc_m_8bit_jpg = ImageOps.grayscale(Image.open('C:/Users/adria/Desktop/Testbilder für Pixel Format/neu/Jpg Bild Expo 0,1.jpg'))
array_jpg=np.array(img_dc_m_8bit_jpg)

img_dc_m_8bit_png = Image.open('C:/Users/adria/Desktop/Testbilder für Pixel Format/neu/Png Bild Expo 0,1.png')
array_png=np.array(img_dc_m_8bit_png)

img_dc_m_8bit_tiff = Image.open('C:/Users/adria/Desktop/Testbilder für Pixel Format/neu/Tiff Bild Expo 0,1.tiff')
array_tiff=np.array(img_dc_m_8bit_tiff)

#Intensität der Bilder bestimmen (also Gray Values addieren)
images=[array_raw,array_bmp,array_jpg,array_png,array_tiff]
Intensities=np.zeros(len(images))

for i in range(0,height):
    for j in range(0,width):
        Intensities[0] += array_raw[i,j]
        Intensities[1] += array_bmp[i,j]
        Intensities[2] += array_jpg[i,j]
        Intensities[3] += array_png[i,j]
        Intensities[4] += array_tiff[i,j]`

thiesmoeller, 2025-01-07:
I would recommend to take a look into matplotlib imshow. 

You write loops over pixels this is extremely slow.
Use numpy operations to work with images
 img.sum() e.g.

Nemarott, 2025-01-08:
> I would recommend to take a look into matplotlib imshow.
> 
> You write loops over pixels this is extremely slow. Use numpy operations to work with images img.sum() e.g.

Thanks for the tipp with ing.sum() that speeds things up. But I still have the problem that  TIFF value is 16 times higher than the RAW value. Is there a "better" solution than just dividing the TIFF value by 16?

thiesmoeller, 2025-01-10:
What is the pixelformat on the camera you used?
issues like what you described, typically comes from different multi-byte layout definitions.

Our cameras output e.g. 12bit data LSB aligned.
So the range is from 0 ... 4095
If the tiff file you created now uses MSB alignment the 12bit pixel values are shifted four bits ( times 16 ) 

So yes, 
when using different file formats and their multi-byte layouts, you have to  check what the values are and adjust accordingly..

```python
import numpy as np
import matplotlib.pyplot as plt

def vectorized_bit_histogram(image: np.ndarray, density: bool = True):
    assert image.dtype == np.uint16
    bit_counts = np.zeros(16, dtype=np.int64)
    for i in range(16):
        bit_counts[i] = np.sum((image & (1 << i)) > 0)
    if density:
        bit_counts = bit_counts / np.sum(bit_counts)
    return bit_counts

def show_bit_histogram(image: np.ndarray):
    assert image.dtype == np.uint16
    bit_counts = vectorized_bit_histogram(image)
    
    plt.figure(figsize=(12, 6))
    plt.bar(np.arange(0, 16), bit_counts)
    
    # Set x-axis labels
    plt.xticks(np.arange(0, 16), [f'$bit_{{{x}}}$' for x in range(16)])
    
    plt.xlabel('Bit Position')
    plt.ylabel('Density')
    plt.title('Bit Histogram')
    plt.show()

# Create a random image with values from 0 to 4095
width, height = 800, 600
random_image = np.random.randint(0, 4096, (height, width), dtype=np.uint16)

show_bit_histogram(random_image)
```

Nemarott, 2025-01-13:
> What is the pixelformat on the camera you used? issues like what you described, typically comes from different multi-byte layout definitions.
> 
I use Mono 12 as the pixelformat. Is there another way to check if the TIFF file is MSB alligned besides adding all grayvalues together add comparing with RAW?

thiesmoeller, 2025-01-13:
To my knowledge tiff has no tag to tell about the alignment.

So you have to do it by hand.

One rough check for a test could be this

```python
import numpy as np
from PIL import Image

def check_tiff_bit_order(filename):
    """
    Checks the bit order (MSB or LSB) of a 16-bit TIFF file using NumPy.

    Args:
        filename: Path to the TIFF file.

    Returns:
        "MSB" if the bit order is Most Significant Bit first,
        "LSB" if the bit order is Least Significant Bit first,
        "Unknown" if the bit order cannot be determined.
    """

    try:
        img = Image.open(filename)

        if img.bits != 16:
            print(f"Warning: '{filename}' is not a 16-bit image.")
            return "Unknown"

        # Extract image data as NumPy array
        data = np.array(img.getdata(band=0), dtype=np.uint16)

        # Calculate sum of high-order and low-order bytes using NumPy
        high_order_sum = np.sum(data >> 8)
        low_order_sum = np.sum(data & 0xFF)

        return "MSB" if high_order_sum > low_order_sum else "LSB"

    except Exception as e:
        print(f"Error processing '{filename}': {e}")
        return "Unknown"

# Example usage
filename = "your_image.tif" 
bit_order = check_tiff_bit_order(filename)
print(f"Bit order of '{filename}': {bit_order}") 
```

Nemarott, 2025-01-14:
Ok thnaks for all the answers.

closed 2025-01-14 by Nemarott