basler/pypylon #805: Camera Link Camera: The attached device does not support grabbing.

universea, 2025-01-07:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

The attached device does not support grabbing.

I can config the camera in Pylon, and I can also use other software (HIKROBOTICS, MVS) to get image. However, it not work on this code generate by pylon.

Code is:

`// Include files to use the pylon API.
#include <pylon/PylonIncludes.h>
#ifdef PYLON_WIN_BUILD
#    include <pylon/PylonGUI.h>
#endif

// Namespace for using pylon objects.
using namespace Pylon;

// Settings to use any camera type.
#include "BaslerCamera.h"
#include "BaslerCameraArray.h"

using namespace Pylon;
using namespace Pylon::BaslerCameraCameraParams_Params;

// Number of images to be grabbed.
static const uint32_t c_countOfImagesToGrab = 10;

// Namespace for using cout.
using namespace std;

int main(int argc, char* argv[])
{
    // The exit code of the sample application.
    int exitCode = 0;

    // Before using any pylon methods, the pylon runtime must be initialized. 
    PylonInitialize();

    try
    {
        // Create an instant camera object with the first found camera device.
        BaslerCamera camera( CTlFactory::GetInstance().CreateFirstDevice());

        // Print the model name of the camera.
        cout << "Using device " << camera.GetDeviceInfo().GetModelName() << endl;
        
        // Open the camera for accessing the parameters.
        camera.Open();

        // Get parameter values
        cout << "camera.AcquisitionFrameCount = " << camera.AcquisitionFrameCount.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AcquisitionFrameRateAbs = " << camera.AcquisitionFrameRateAbs.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AcquisitionFrameRateEnable = " << camera.AcquisitionFrameRateEnable.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AcquisitionMode = " << camera.AcquisitionMode.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AcquisitionStart = " << camera.AcquisitionStart.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AcquisitionStatus = " << camera.AcquisitionStatus.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AcquisitionStatusSelector = " << camera.AcquisitionStatusSelector.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AcquisitionStop = " << camera.AcquisitionStop.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoExposureTimeAbsLowerLimit = " << camera.AutoExposureTimeAbsLowerLimit.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoExposureTimeAbsUpperLimit = " << camera.AutoExposureTimeAbsUpperLimit.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionAOIHeight = " << camera.AutoFunctionAOIHeight.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionAOIOffsetX = " << camera.AutoFunctionAOIOffsetX.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionAOIOffsetY = " << camera.AutoFunctionAOIOffsetY.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionAOISelector = " << camera.AutoFunctionAOISelector.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionAOIUsageIntensity = " << camera.AutoFunctionAOIUsageIntensity.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionAOIUsageRedLightCorrection = " << camera.AutoFunctionAOIUsageRedLightCorrection.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionAOIUsageWhiteBalance = " << camera.AutoFunctionAOIUsageWhiteBalance.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionAOIWidth = " << camera.AutoFunctionAOIWidth.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoFunctionProfile = " << camera.AutoFunctionProfile.ToStringOrDefault("<not readable>") << std::endl;
        cout << "camera.AutoGainRawLowerLimit = " << camera.AutoGainRawLowerLimit.ToStringOrDefault("<not readable>") << std::endl;
[... 384 lines cut ...]
camera.SyncUserOutputValue = 0
camera.SyncUserOutputValueAll = 0
camera.TestImageSelector = Off
camera.TriggerActivation = RisingEdge
camera.TriggerDelayAbs = 0
camera.TriggerMode = Off
camera.TriggerSelector = FrameStart
camera.TriggerSoftware = <not readable>
camera.TriggerSource = CC1
camera.UserDefinedValue = 0
camera.UserDefinedValueSelector = Value1
camera.UserOutputSelector = UserOutputLine1
camera.UserOutputValue = 0
camera.UserOutputValueAll = 0
camera.UserSetDefaultSelector = Default
camera.UserSetLoad = <not readable>
camera.UserSetSave = <not readable>
camera.UserSetSelector = Default
camera.Width = 2044
camera.WidthMax = 2048
An exception occurred.
The attached device does not support grabbing.

Press Enter to exit.`
```

SMA2016a, 2025-01-10:
you need to use the sdk of frame grabber manufacture to grab images.  Pylon SDK does not support streaming with CL Interface

SMA2016a, 2025-01-10:
by the way, in this forum we handle only the questions related to pypylon only.

closed 2025-01-10 by SMA2016a