basler/pypylon #806: BUG: hasattr not working with InstantCamera

andreas-h-sie, 2025-01-08:
### Describe the issue:

`hasattr` does not seem to work with `InstantCamera`.

Internally Python implements `hasattr` with a `getattr` and checks wether an AttributeError is raised or not. However, pypylon seems to raise an LogicalErrorException which is not according to hasattr expectation.

As a workaround I use the following

```python 
def pylon_hasattr(obj, attr: str) -> bool:
    return attr in dir(obj)
```

### Reproduce the code example:

```python
from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

hasattr(camera,"GevIEEE1588")  # this raises _genicam.LogicalErrorException
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

a2A2590-22gmPRO

Linux 6.1

### Runtime information:

```shell
python: 3.11.6 (main, Nov 29 2023, 04:47:02) [GCC 10.2.1 20210110]
platform: linux/x86_64/6.1.102-cip26-rt14+ind3
pypylon: 1.9.0 / 6.3.0.18933
```

thiesmoeller, 2025-01-08:
Thank you for reporting this issue. 

The problem arises because the hasattr function in Python expects an AttributeError when an attribute is not present, while pypylon currently raises a _genicam.LogicalErrorException. 
This is the exception that the underlying genicam model raises for missing nodes.

Code is here https://github.com/basler/pypylon/blob/e58a7355f662369eb3a70dffcd7b9716590eb589/src/pylon/InstantCamera.i#L40

To address this, we propose the following solution that maintains backward compatibility while providing a path for future improvements.

---

Solution Overview

1. Introduce a Strict Mode Flag

A new flag, strict_attribute_error, will be introduced to control how missing attributes are handled.

By default, the behavior will remain the same (raising _genicam.LogicalErrorException) to avoid breaking existing code.

When strict_attribute_error is enabled, the library will raise AttributeError instead, aligning with Python's conventions.

2. Per-Instance and Global Configuration

Global Setting: Developers can enable strict mode globally for all objects in the library.

Per-Instance Setting: Strict mode can be enabled individually for specific objects (e.g., InstantCamera) to allow finer control.

3. Future Breaking Change (Next Major Version):

In the next major version release, the library will fully transition to raising AttributeError for missing attributes, removing _genicam.LogicalErrorException from this context. The strict mode flag will then be deprecated.

Please share your view on this.

andreas-h-sie, 2025-01-08:
Changing the behavior and becoming more conforming to Python standards in the next major relase sounds like a good plan.

Not sure if a by-instance "strict mode" is really worth it. Maybe not even a global strict mode at all. Good would be in any case to deprecate the current behavior (raise LogicError). 

If you make a change in the next major release perhaps a global legacy mode flag (the inverse of strict mode) (which restores the old behavior) makes sense, in case some existing code relies on the old behavior. 

In the meantime you perhaps could document the not-working hasattr as known limitation?