basler/pypylon #807: BUG: Can change "PTP Enable" in pylonviewer, but can't set `PtpEnable` with pypylon

SamDM, 2025-01-09:
### Describe the issue:

I am following [this][1] documentation in attempt to clock-sync multiple basler cameras.
However, the code instructions I find there don't seem to work.

This script correctly prints my camera serial numbers:
```
['Basler a2A4200-12gcBAS (40400219)', 'Basler a2A4200-12gcBAS (40400220)', 'Basler a2A4200-12gcBAS (40400221)', 'Basler a2A4200-12gcBAS (40400222)']
```

But then raises an error.
The odd thing is that in the pylonviewer, I can check/uncheck the `Enable Ptp` value just fine.

Am I missing a step?

  [1]: https://docs.baslerweb.com/precision-time-protocol#enabling-ptp-clock-synchronization

### Reproduce the code example:

```python
from pypylon import pylon

_tlf: pylon.TlFactory = pylon.TlFactory.GetInstance()

devices: list[pylon.DeviceInfo] = list(filter(lambda d: d.GetModelName() == "a2A4200-12gcBAS", _tlf.EnumerateDevices()))
print([d.GetFriendlyName() for d in devices])
cam_array: pylon.InstantCameraArray = pylon.InstantCameraArray(len(devices))
for device, cam in zip(devices, cam_array, strict=True):
    cam.Attach(_tlf.CreateDevice(device))
    cam.Open()

for cam in cam_array:
    assert cam.IsOpen()
    cam.PtpEnable.Value = True
```

### Error message:

```shell
Traceback (most recent call last):
  File "/home/samdm/Safe/0041-VFE/rt-vfe/packages/vfe_rt_plugin_basler/scripts/ptp_stackoverflow.py", line 14, in <module>
    cam.PtpEnable.Value = True
    ^^^^^^^^^^^^^^^^^^^
  File "/home/samdm/Safe/0041-VFE/rt-vfe/packages/vfe_rt_plugin_basler/.venv/lib/python3.12/site-packages/pypylon/genicam.py", line 2073, in SetValue
    return _genicam.IBoolean_SetValue(self, Value, Verify)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
_genicam.AccessException: Node is not writable. : AccessException thrown in node 'PtpEnable' while calling 'PtpEnable.SetValue()' (file 'BooleanT.h', line 61)
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

System:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04.1 LTS
Release:        24.04
Codename:       noble
$ uname -a
Linux VFE-server 6.8.0-51-generic #52-Ubuntu SMP PREEMPT_DYNAMIC Thu Dec  5 13:09:44 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```
62Gb RAM

Network interface:
```
$ ifconfig
enp10s0f0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9000
        inet 192.168.3.2  netmask 255.255.255.0  broadcast 192.168.3.255
        inet6 fe80::eb2b:5985:8f31:fb38  prefixlen 64  scopeid 0x20<link>
        ether a0:36:9f:bc:2c:39  txqueuelen 1000  (Ethernet)
        RX packets 10430  bytes 7346788 (7.3 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6146  bytes 398366 (398.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 19  memory 0x886a0000-886c0000  

enp10s0f1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9000
        inet 192.168.6.2  netmask 255.255.255.0  broadcast 192.168.6.255
        inet6 fe80::9240:4b72:1ee0:e07a  prefixlen 64  scopeid 0x20<link>
        ether a0:36:9f:bc:2c:38  txqueuelen 1000  (Ethernet)
        RX packets 6876  bytes 2304518 (2.3 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4280  bytes 279102 (279.1 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 18  memory 0x88640000-88660000  

enp11s0f0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9000
        inet 192.168.7.2  netmask 255.255.255.0  broadcast 192.168.7.255
        inet6 fe80::cc2c:6d8e:268d:1bc8  prefixlen 64  scopeid 0x20<link>
        ether a0:36:9f:bc:2c:3b  txqueuelen 1000  (Ethernet)
        RX packets 9196  bytes 2644918 (2.6 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6577  bytes 426082 (426.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 17  memory 0x885a0000-885c0000  

enp11s0f1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 9000
        inet 192.168.8.2  netmask 255.255.255.0  broadcast 192.168.8.255
        inet6 fe80::9c31:ab63:c840:19fd  prefixlen 64  scopeid 0x20<link>
        ether a0:36:9f:bc:2c:3a  txqueuelen 1000  (Ethernet)
        RX packets 8515  bytes 2615264 (2.6 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 5882  bytes 381412 (381.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  memory 0x88540000-88560000  
```

```
$ lshw -C network
  *-network:0
       description: Ethernet interface
       product: 82571EB/82571GB Gigabit Ethernet Controller (Copper)
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:0a:00.0
       logical name: enp10s0f0
       version: 06
       serial: a0:36:9f:bc:2c:39
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 32 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list rom ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=e1000e driverversion=6.8.0-51-generic duplex=full firmware=5.10-2 ip=192.168.3.2 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
       resources: irq:158 memory:886a0000-886bffff memory:88680000-8869ffff ioport:4020(size=32) memory:88660000-8867ffff
  *-network:1
       description: Ethernet interface
       product: 82571EB/82571GB Gigabit Ethernet Controller (Copper)
       vendor: Intel Corporation
       physical id: 0.1
       bus info: pci@0000:0a:00.1
       logical name: enp10s0f1
       version: 06
       serial: a0:36:9f:bc:2c:38
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 32 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list rom ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=e1000e driverversion=6.8.0-51-generic duplex=full firmware=5.10-2 ip=192.168.6.2 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
       resources: irq:177 memory:88640000-8865ffff memory:88620000-8863ffff ioport:4000(size=32) memory:88600000-8861ffff
  *-network:0
       description: Ethernet interface
       product: 82571EB/82571GB Gigabit Ethernet Controller (Copper)
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:0b:00.0
       logical name: enp11s0f0
       version: 06
       serial: a0:36:9f:bc:2c:3b
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 32 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list rom ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=e1000e driverversion=6.8.0-51-generic duplex=full firmware=5.10-2 ip=192.168.7.2 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
       resources: irq:178 memory:885a0000-885bffff memory:88580000-8859ffff ioport:3020(size=32) memory:88560000-8857ffff
  *-network:1
       description: Ethernet interface
       product: 82571EB/82571GB Gigabit Ethernet Controller (Copper)
       vendor: Intel Corporation
       physical id: 0.1
       bus info: pci@0000:0b:00.1
       logical name: enp11s0f1
       version: 06
       serial: a0:36:9f:bc:2c:3a
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 32 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list rom ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=e1000e driverversion=6.8.0-51-generic duplex=full firmware=5.10-2 ip=192.168.8.2 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
       resources: irq:179 memory:88540000-8855ffff memory:88520000-8853ffff ioport:3000(size=32) memory:88500000-8851ffff
```

### Runtime information:

```shell
python: 3.12.3 (main, Nov  6 2024, 18:32:19) [GCC 13.2.0]
platform: linux/x86_64/6.8.0-51-generic
pypylon: 4.1.0 / 9.0.3.215
```

SamDM, 2025-01-09:
The issue seems to happen with `BslPtpManagementEnable`

```
_genicam.AccessException: Node is not writable. : AccessException thrown in node 'BslPtpManagementEnable' while calling 'BslPtpManagementEnable.SetValue()' (file 'BooleanT.h', line 61)
```

thiesmoeller, 2025-01-09:
At the end of the document you reference the code to enable ptp is shown.

SamDM, 2025-01-10:
After using the GUI for trying out different combination of the settings mentioned in the docs, suddenly it works in the code as well. Now, I can't seem to replicate the discrepancy in behavior anymore between the GUI and the pylon SDK. I.e. the same combination of settings that enable PtP in the GUI also enables setting it with the SDK.

I am trying to figure out now what went wrong before, since now I'm failing to reproduce this discrepant behavior.
There must have been some state in the camera that was blocking me when trying this before, which is not longer the case now, but I can't figure out what it was.

Anyway, thanks for the feedback.
I'll close this given that it "works" now.
If I ever find out, I'll post it here in case someone else runs into the same trouble.

closed 2025-01-10 by SamDM

thiesmoeller, 2025-01-10:
Hi @SamDM,

the cameras keep state, even if you close the connection.

To prevent what you described, that operating the camera in PylonViewer 
or running another script before,
it is recommended to always start with a known state

```python
import pypylon.pylon as py
tlf = py.TlFactory.GetInstance()

cam = py.InstantCamera(tlf.CreateFirstDevice())
cam.Open()
# set the camera to factory settings
# to start with the default state
cam.UserSetSelector.Value = "Default"
cam.UserSetLoad.Execute()

# your application code...

cam.Close()
```