basler/pypylon #808: How to display intensive image by opencv

shawn2766, 2025-01-10:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi, I use opencv.py in the sample code to display images from my blaze101 camera, but what is displayed is the depth image. How can I get the intensity image and display it through opencv? Is there any relevant sample code? Thanks in advance.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

win11

### Camera(s) used

basler blaze-101

### Runtime information:

```shell
python3.10, pypylon: 4.0.0
```

SMA2016a, 2025-01-14:
you need to split the grab buffer in 2 parts.  and create simply 2 images.

dowolters, 2025-02-13:
If you install the current version of the [pylon Supplementary Package for ToF Camera / blaze 1.7.2](https://www.baslerweb.com/en/downloads/software/?downloadCategory.values.label.data=pylon+Supplementary+Package), you will find code samples for using the blaze camera with pypylon in the ‘C:\Program Files\Basler\pylon 8\Development\Samples\blaze\Python\pypylon\’ directory. These code samples show the configuration of the camera and also how to display the intensity image, depth image and the point cloud.