basler/pypylon #809: CameraEventHandler and record/show video

PatrikHernandez, 2025-01-12:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,

I am currently working in an application where we use a Basler camera to capture images (hw triggering, frame-rate = 35 Hz), show them in real time and save them as a video as well. Unfortunately, pylon viewer cannot be used for such tasks due to some hard constraints from our side, and we rather have to implement the whole procedure in python. In the implementation process, I faced up 3 issues, for which I would kindly ask your feedback:

## Question 1 (acquiring):

I am using a camera event (frameStartEvent) to modify the duration of Timer1 each time a picture is acquired (in a background loop; camera thread) and the ExposureStart parameter is precisely dependent on Timer1 (there is a delay between the frameStart and exposureStart). However, the modification only affects the next image to be acquired. Both frameStart and Timer1 are triggered by the same hw trigger (so virtually at the same time).  The sequence is basically:

1. hw trigger for frame N (FrameStart and Timer1 get started) 
2. frameStartEvent for frame N where we set timer1 duration for frame N + 1
3. Exposure start for frame N, based on Timer1 duration set in frame N-1
4. Exposure end for frame N
5. hw trigger for frame N + 1 (...)

I have seen however that the change of the Timer1 duration set at frame N (while acquiring frame N) will not always be the value which I read (and is used) from ``camera.TimerDuration.Value`` at frame N+1 . In other words, it seems that sometimes (like 0-6 times in 1000 frames) I am not able to change Timer1 in the current frame so that it gets/uses these new parameters in the next frame. The code I have in frameStartEvent is basically just setting the timer duration (no heavy processing task). The fact that this occurrence seems random somehow tells me that it could be a performance issue and I believe there is no way to "block" the camera thread and force it to execute the frameStartEvent for the current frame, before going to the next frame. Or is it always executing it but sometimes Timer1 might not be "available" for changes (since it is being executed while its parameters are being changed, even if only to take effect at the next frame)? Is my understanding correct? We are essentially speaking about a runtime change of the camera parameters. Unfortunately, the sequencer feature can not be used for what I want.

## Question 2 (video write):

I am currently using opencv to make the video from the frames acquired, but my issue is that the resulting video is always much bigger than a similar one recorded with PylonViewer, using apparently similar codecs (video format is always .mp4 and codec mp4v in both cases). Typical values are 400 MB with opencv and 100 MB for PylonViewer (with quality set at 90 %) for a 1000 frames video, with the camera pixel format as BayerRG8 (and full resolution). Perhaps due to this size difference, opencv also struggles to write the video as fast as the frames are acquired (I am using a queue, and I see that images get accumulated over time, while the recording buffer in PylonViewer keeps at 0 %; I could also test it with CVideoWritter in pylon C++ and observe the same behavior as in PylonViewer, as expected).
Question: what would be the set of conditions to use in opencv in order to record a video with the same specs as in Pylon Viewer (or what would be the way to use in python as close as CVideoWritter is doing)? Note: I was not able to see any python wrapper of the CVideoWritter class in C++, can you confirm it? I have seen also the suggestion made here https://github.com/basler/pypylon/issues/113#issuecomment-543774545, with ffmpeg in a subprocess, but it seems not to be working (I get a weird video, probably due to some pixel format issue; I am also not sure about the performance of this method vs opencv)
The way I am doing it (based on some of your tutorials):
```python

vidw = cv2.VideoWriter("video.mp4", cv2.VideoWriter_fourcc(*'mp4v'), 35, [2048, 2448])
converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned`

while loop:
    img = converter.Convert(grabResult).GetArray()
    vidw.write(img)
```

## Question 3 (image show):

Lastly, is there any way that I could use ``pylon.PylonImageWindow()`` (or any similar pylon method) to display any generic image (let´s say a RGB png picture which I load from the disc or an equivalent numpy array)? I have been trying to pass different things to ``imageWindow.SetImage()``, but it seems it only accepts a ptr to a grab result. The point here is that I dont want to display the grabresult as it is, but after some image manipulation (for example, two grabresults concatenated side by side or one grabresult where I inserted some text). I am currently using opencv to manipulate & show the pictures in real time (like a video), but the performance is also not the best (I have seen better performance of PylonImageWindow in a case where I could use/compare both of them). The target would be to play the frames in real time as fast as possible (best performance).
The way I am doing it (based on some of your tutorials):
```python

converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned`
cv2.namedWindow('Win1')
while loop:
    img = converter.Convert(grabResult).GetArray()
    cv2.putText(...) # Add some text to img
    cv2.imshow('Win1', img)
    key=cv2.waitKey(1) 
    if key == 27: 
        break
```

Thank you for your time.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

X86
Win10
USB3.0

### Camera(s) used

Basler ace 2 R
a2A2448-75ucPRO

### Runtime information:

```shell
python: 3.12.6
platform: win32/AMD64/10
pypylon: 4.0.0/8.0.0.10
```

thiesmoeller, 2025-01-14:
Hi @PatrikHernandez,

the camera interaction will be tricky, as you run on windows, which is not a real time operating system.
So reactiong to an event of the camera and modifying a value **just in time** can  ( and will, as you decribed ) fail.

Can you describe what is the problem you want to solve with your code, especially as you have already checked the camera 
sequencer?

PatrikHernandez, 2025-01-14:
Hi @thiesmoeller,

I agree that it is a tricky task, even if the parameters that we change while a frame is being taken are only to be effective for the next frame.

The application is basically an inspection line where we have several variants of a product. The products arrive at a more or less constant rate, in a "random" order (we cannot control the order of arrival, but I know, before imaging - like 5 frames before - , which variant is going to pass below the camera). However, each variant is to be inspected (imaged) at a different position. This means constant hw triggering, but variable exposure **start** (exposure time is always the same, ie end-start = @) over time (that we control with timers because the max trigger delay value of 10 ms is not enough). Sequencer allows to engage the exposure time, but that is not useful in our case. Besides, since the order (sequence) is unknown at the time of setting up the sequencer, that wouldnt work either.

Would you have suggestions for the other two questions as well?

thiesmoeller, 2025-01-15:
For the video window just use opencv.

```python

while camera.IsGrabbing():
    with camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grab_result:
        if grab_result.GrabSucceeded():
            img = grab_result.Array
             img= cv2.cvtColor(img ,  < camera format to BGR>)
            
            # Display the image
            cv2.imshow('Basler Camera', img)

            # Break on 'q' key press
            if cv2.waitKey(1) & 0xFF == ord('q'):
                break

cv2.destroyAllWindows()
```

thiesmoeller, 2025-01-15:
For the special capturing requirements you have typically you would use a higher resolution camera with a wider field of view and cut your different position image after capturing

Or have the PLC that generates the triggers implement the special delayed trigger per object that you know upfront.

Or as a third option please contact your local sales/support  for a camera customization that could implement a special FIFO mode for you to queue in TriggerDelay setting for the next frames ( with the timing you need )

What you do will always be asynchronous to the capture process of the camera. You should at least increase the scheduling priority of your python process to make this a little bit more deterministic

closed 2025-01-16 by PatrikHernandez