basler/pypylon #810: How to solve "Node not existing (file 'genicam_wrap.cpp', line xxxxx)"

cowkim937, 2025-01-15:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I'm trying to develop a camera app using PySide2 with MVC pattern.

The problem is that when I open the camera, the camera gives me a warning: "Node does not exist (file 'genicam_wrap.cpp')".

The device parameter in connect method from device_list property.

Why it didn't open to camera?

Below is define library and class
```
from pypylon import pylon

class Basler:
    def __init__(self):
        self._tlf = None
        self._camera = None
        self._converter = None
```

And used to pypylon code is below
```
@property
def device_list(self):
    self._tlf = pylon.TlFactory.GetInstance()
    return self._tlf.EnumerateDevices()

def connect(self, device):
    self._camera = pylon.InstantCamera(self._tlf.CreateDevice(device))
    try:
        self._camera.Open()
    except Exception as e:
        print(e)
        return False
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Camera: acA3088-57uc 
Connection type: USB3

### Camera(s) used

Basler acA3088-57uc (25020872)
107403-16;U;acA3088_57uc;V1.5-3;1

### Runtime information:

```shell
python: 3.9.13 (tags/v3.9.13:6de2ca5, May 17 2022, 16:36:42) [MSC v.1929 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 4.1.0 / 9.0.3.215
```

cowkim937, 2025-01-16:
Ah, the problem was caused by incorrectly accessing the camera settings option from another line.

In particular, this happen occurs when touching the FPS or Gain settings.

like,
```
self._camera.Gain = 10.0
```

closed 2025-01-16 by cowkim937

thiesmoeller, 2025-01-16:
The pattern you use is deprecated.
Please use 

`self._camera.Gain.Value = 10.0`