basler/pypylon #812: Reopen connection after process termination

Kukuevm, 2025-01-21:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am looking for a way to reopen connection when the program is closed or crashes without closing the camera session.

The following code establishes the connection every second run if the previous one ended up with an error "The device is controlled by another application. "0xE1018006""
```
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
```

However, I would like to make the program to do a few re-connection attempts like shown below:
```
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

status = False
while True:
    try:
        if status == False:
            camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 
            status = True
    except Exception:
        time.sleep(4) # I assumed that after 3 seconds the connection should be closed by the camera but it is not
        continue
```
But the code above always fails to exception.

There seems to be a solution for C++ called PylonGigEConnectionGuard to resolve this particular issue but I cannot find any alternatives for Python.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Windows 10 x64

### Camera(s) used

BaslerCamera a2A 1920-51gcPRO (40429581) 
p=vg4p_imx392c_pro/s=r/v=2.4.2/i=8208.11/h=02e36ab

### Runtime information:

```shell
python: 3.12.4 | packaged by Anaconda, Inc. | (main, Jun 18 2024, 15:03:56) [MSC v.1929 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 4.0.0 / 8.0.0.10
```

Kukuevm, 2025-02-08:
The Heartbeat Timeout is set to 3 seconds but the script still cannot re-connect after this time when a "try" cycle is running.
The only options to open the connection again are either to power cycle the camera or re-run the script with the following two lines twice.
```
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
```

What I cannot get is why re-running these two lines in a "try" cycle (even with a 5-second timeout) does not reset the connection but re-running the script does.

Another interesting thing is if the script is killed by a task manager, the connection can be opened right away even though it has not been properly closed. So the problem occurs only when the script crashes.

thiesmoeller, 2025-02-08:
It is the inverse that you have to do.
The heartbeat timeout defines the time the camera will keep a connection open without getting any requests from your application.

If it works with Closing and "killing" your application: then the destructors of the instantcamera are still properly called and the connection is properly closed