basler/pypylon #814: help for the get ip address of blaze101

fkdlqj13245, 2025-01-23:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Can you help to check blaze101's ip address?
If i try to get the camera's ip address, the ace camera can get the ip address.
But Blaze couldn't get the address.

I tried with below code.

#Code

tl_factory = pylon.TlFactory.GetInstance()

devs_info = tl_factory.EnumerateDevices()

for dev_info in devs_info:

# if dev_info.GetDeviceClass() == 'BaslerGigE':

    cam_info = dev_info

print(

        dev_info.GetDeviceClass() , ', ',

            cam_info.GetModelName(), ', ',

            cam_info.GetIpAddress(), ', ',

            cam_info.GetMacAddress()

    )

#result
BaslerGigE ,  acA1300-75gc ,  192.168.91.176 ,  0030534D7D7B
BaslerGTC/Basler/GenTL_Producer_for_Basler_blaze_101_cameras ,  blaze-101 ,  N/A ,  N/A

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

OS: Ubuntu 20.04(kernel version: 5.15.0-94-generic)

GPU: RTX3060
PYTHON: 3.8.10

### Camera(s) used

acA1300-75gc, blaze 101

### Runtime information:

```shell
pypylon version: 3.0.1 ( also tested 4.0.0)

pylon viewer: 7.4 64bit
```

alexanderhappe, 2025-01-27:
Unfortunately the IP address information is currently not available in the pylon device enumeration info objects for Basler blaze cameras.
I assume you want to use the IP address for camera identifcation? Is using the serial number or the information from the Device User ID feature an alternative?
If you really require the IP address, you have to open the device first and query the IP address configuration from the device.

Since the missing IP address information is not a pypylon specific issue but a general Basler pylon restriction, feel free to open a support case for Basler pylon on the Basler website.

alexanderhappe, 2025-02-12:
The issue has been addressed by the pylon development team. With the upcoming pypylon release it will be possible to retrieve the IP address, subnet mask and MAC address from a device info object returned by the device enumeration.