basler/pypylon #815: Grab failed. Code: 3791651074 Description: The buffer was cancelled.

Satge96, 2025-01-24:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

```
import warnings

from pypylon import pylon

def main():
    print("Hello from basler-camera-problems!")
    tlf = pylon.TlFactory.GetInstance()
    devices = tlf.EnumerateDevices()
    n_cameras = len(devices)
    camera_array = pylon.InstantCameraArray(n_cameras)

    camera_indices = [0, 1]
    for idx, (camera_idx, camera) in enumerate(zip(camera_indices, camera_array)):
        camera.Attach(tlf.CreateDevice(devices[idx]))
        camera.SetCameraContext(camera_idx)
    camera_array.Open()
    for cam in camera_array:
        cam.AcquisitionFrameRateEnable.SetValue(True)
        cam.AcquisitionFrameRate.SetValue(20)
        cam.TriggerSelector.SetValue("FrameStart")
        cam.TriggerMode.SetValue("Off")
        cam.AcquisitionMode.SetValue("Continuous")
        cam.Width.SetValue(cam.SensorWidth.GetValue())
        cam.Height.SetValue(cam.SensorHeight.GetValue())
        cam.OffsetX.SetValue(0)
        cam.OffsetY.SetValue(0)
        cam.ReverseX.SetValue(False)
        cam.ReverseY.SetValue(False)

    camera_array.StartGrabbing(
        pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByUser
    )
    while True:
        with camera_array.RetrieveResult(
            5000, pylon.TimeoutHandling_ThrowException
        ) as grab_result:
            if grab_result.GrabSucceeded():
                if pylon.PayloadType_Image != grab_result.PayloadType:
                    warnings.warn("Unexpected payload type received")
                    continue
                camera_idx = grab_result.GetCameraContext()
                frame_counter = grab_result.ImageNumber
                grab_result.Release()
            else:
                msg = f" Grab failed. Code: {grab_result.GetErrorCode()}"
                msg += f" Description: {grab_result.GetErrorDescription()}"
                print(msg)

if __name__ == "__main__":
    main()
```

After a while, the code above results in the following:
Grab failed. Code: 3791651074 Description: The buffer was cancelled.
(A while being around 30min)

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup used

X86_64
Ubuntu 24.04.1 LTS
32.0 GiB RAM

### Camera(s) used

Two a2A4200 - 40umBAS

### Runtime information:

```shell
python: 3.11.11 (main, Dec  4 2024, 08:55:08) [GCC 13.2.0]
platform: linux/x86_64/6.8.0-51-generic
pypylon: 4.1.0 / 9.0.3.215
```

thiesmoeller, 2025-01-24:
First comment:
You do a double free of the grab result.

The with... Handler will automatically release on exit of the context

Satge96, 2025-01-30:
That seemed to fx my problem in the minimal project. I will try it in the bigger one.

Satge96, 2025-03-07:
I switched from multiprocessing to only multithreading in the bigger project. That fixed most of my problems. However, I'm not sure if the buffer set with MaxNumBuffer is released correctly. Can you tell me what the correct way to release this buffer is? I have a multicamera setup and call camera_array.Close() but in the [multicam example](https://github.com/basler/pypylon/blob/master/samples/grabmultiplecameras.py) you do nothing and I wonder if something else is needed to release the buffer.