basler/pypylon #819: Running Control and Monitor Mode on the Same Computer Fails Due to Socket Error

ocesnil, 2025-02-07:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

We want to run two applications on the same computer, one in control mode (master) and the other in monitor mode (listener). To achieve this, we created two separate scripts and ran them in different terminals. The control mode script successfully starts grabbing images, but when we launch the monitor mode script, it fails with an error stating that the socket address is already in use.
The Pylon documentation suggests running control mode and monitor mode on two separate computers, but we need to run both on the same machine. [https://docs.baslerweb.com/pylonapi/cpp/sample_code#grab_multicast](url) We tested this setup with and without multicast enabled, but the issue persists in both cases. The error message indicates that the monitor mode script cannot open the stream grabber because the socket is already being used by the control mode script:

_genicam.RuntimeException: Failed to open stream grabber: Only one usage of each socket address (protocol/network address/port) is normal permitted

Is there a way to run both control mode and monitor mode on the same computer without encountering this socket conflict? 

master_mode.py
```
from pypylon import pylon

tl_factory = pylon.TlFactory.GetInstance()
camera = pylon.InstantCamera(tl_factory.CreateFirstDevice())

camera.Open()
#camera.StreamGrabber.TransmissionType.Value = "Multicast" 
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

while camera.IsGrabbing():
    try:
        grab_result = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        if grab_result.GrabSucceeded():
            #print(f"Multicast Frame Available ({grab_result.Width}x{grab_result.Height})")
            grab_result.Release()
    except KeyboardInterrupt:
        break

camera.Close() 
```

monitor_mode.py
```
from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.MonitorModeActive = True
camera.RegisterConfiguration(ConfigurationEventPrinter(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)

camera.Open()
#camera.StreamGrabber.TransmissionType.Value = "Multicast" 
camera.StartGrabbing()

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        print("Success")
    grabResult.Release()

camera.Close()
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Network Setup: Wired, GigE

### Camera(s) used

Camera Model: Basler a2A1920-165g5c

### Runtime information:

```shell
Python 3.11
pylon 8.0.2
```

SMA2016a, 2025-02-09:
try to add camera.RegisterConfiguration(py.ConfigurationEventHandler(), py.RegistrationMode_ReplaceAll, py.Cleanup_Delete)

ocesnil, 2025-02-11:
Thank you for your suggestion. I updated my script to include the suggested ConfigurationEventHandler(), but unfortunately, the error remains. Here’s the updated script I’m using:

monitor_mode.py

```
from pypylon import pylon

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.MonitorModeActive = True
camera.RegisterConfiguration(ConfigurationEventHandler(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete) 

camera.Open()
# camera.StreamGrabber.TransmissionType.Value = "Multicast" 
camera.StartGrabbing()

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

    if grabResult.GrabSucceeded():
        # Access the image data
        print("Success")
    grabResult.Release()

camera.Close()
```

However, I still get the following error:

_genicam.RuntimeException: Failed to open stream grabber: Only one usage of each socket address (protocol/network address/port) is normal permitted

Would you have any insights into what could still be causing this?

thiesmoeller, 2025-02-11:
Monitor mode works by configuring the camera into multicast transmission.
As you are on the same host, the host will receive one packet, that will get duplicated by the operating system and then our drivers will have to decode the gigevision traffic once per application.

This is not efficient.

Why not transferring the final image data between the two apps?

Regarding the error you describe: You havn't written the operating system you use, which is relevant to solve this..
As written above, this is not a recommended operating mode, so we have to check what prevents the port re-use.

ocesnil, 2025-02-11:
The operating system I use is Windows 10 Pro Verion 22H2 (Build 19045.5371)

ocesnil, 2025-02-12:
> Monitor mode works by configuring the camera into multicast transmission. As you are on the same host, the host will receive one packet, that will get duplicated by the operating system and then our [...]

Unfortunately it is not the most efficient way. The main reason that we decided to use multicast is that we want to use pylon viewer simultaneously during image processing by our machine learning application.

thiesmoeller, 2025-02-12:
Can you test to set the driver type to SocketDriver on both pylonviewer and your python application?

in your code before `cam.StartGrabbing(...)` select the socket driver implementation: ` cam.StreamGrabber.Type.SetValue("SocketDriver")`

and in the pylonviewer:

[image]

thiesmoeller, 2025-02-14:
The filter driver interface to userland is full images in the buffer queue.
For your use case the driver would have to create copies of incoming images into the monitor buffer queues.
We simply haven't implemented this, because typically you implement this at a higher layer. 
This has the benefit that it works also for other camera interfaces beside GEV.

But we take your use case of using the pylon viewer as a tuning / control tool and operating the business logic in another process as valuable input.