basler/pypylon #821: Pixel Format Conversion Error in Multi-Camera Setup for a2A2448-105g5cBAS

shimens, 2025-02-17:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am experiencing an issue in a multi-camera environment and would appreciate your assistance.

Issue Description)

Single Camera Operation
- When running the a2A2448-105g5cBAS camera on its own, it works properly with either the mono8 or bayerrg8 pixel format.

Multi-Camera Operation
- When I operate the a2A2448-105g5cBAS camera alongside the a2A640-240gmSWIR and a2A2840-67g5mUV cameras, the a2A2448-105g5cBAS camera outputs an error: “pixel format currently not supported.”
- As a workaround, I switched the a2A2448-105g5cBAS camera’s pixel format to bayerrg12, and it then functions normally with the two additional cameras.
- However, if I attempt to change the pixel format from bayerrg12 to mono8 during runtime, a “grab fail” error occurs, and the camera stops working as expected.

Questions
- Although the a2A2448-105g5cBAS camera’s specifications indicate support for mono8 and bayerrg8 formats, why is the camera failing to operate correctly in these modes when used in a multi-camera setup?
- Could internal processing pipelines, bandwidth limitations, or driver/firmware issues in a multi-camera configuration be contributing to this problem?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

x64

### Camera(s) used

a2A2448-105g5cBAS
a2A640-240gmSWIR
a2A2840-67g5mUV

### Runtime information:

```shell
Pixel format currently not supported 

Grab Failed
```

thiesmoeller, 2025-02-17:
Are the three cameras properly capture images in parallel in your required video modes when you use pylonviewer?

thiesmoeller, 2025-02-17:
The image sensor of the a2A2448-105g5c can be read out in 10bit and 12bit mode. 
When you enable MONO8 output per default the sensor is switched to the much faster 10bit mode.
->you overrun your system. Quick fix could be: 

`PixelFormat -> Mono8`
`BslSensorBitDepthMode -> Manual`
`BslSensorBitDepth -> Bpp12`

But you should adjust the bandwidth of the cameras connected to your system setup: 
As you haven't written anything about your system setup, it is hard to help here...

https://docs.baslerweb.com/network-configuration-(gige-cameras)
https://docs.baslerweb.com/overview-of-the-pylon-gige-configurator
https://docs.baslerweb.com/bandwidth-manager

And the error you get from pypylon "pixel format not supported" is typically only visible, when you try to work with an incomplete grab.

Can you check your code, if you properly check the grab-status before using the grab result?
https://github.com/basler/pypylon/blob/e58a7355f662369eb3a70dffcd7b9716590eb589/samples/grab.py#L54

shimens, 2025-02-18:
I'm still experiencing issues with my multi-camera setup and would appreciate more detailed assistance. Here are the details of my system configuration and the relevant code I'm using:

1. Hardware Info
OS: Windows 11 Pro
CPU: Intel® Core™ Ultra 7 155H 1.40 GHz
Memory: 32GB RAM

2. Camera Info
Models:
a2A2448-105g5c
a2A640-240gmSWIR
a2A2840-67g5mUV
Connection: All cameras are connected via a network switch.

3. Network Info
Network Adapter: Intel Ethernet Connection (18) I219-V
Network Switch: 10-Port Gigabit Desktop Switch with 8-Port PoE+

4. Code Snippet
Below is the key part of my code:

python

```
def connect_camera():
    global camera_connected, start_flags, cameras, camera_model_names, trigger_ready

    try:
        print("Connecting to cameras...")
        time.sleep(3)  # Simulate connection delay

        tl_factory = pylon.TlFactory.GetInstance()
        devices = tl_factory.EnumerateDevices()
        device_count = len(devices)
        print(f"Number of camera devices found: {device_count}")

        # Print detailed device information (for debugging)
        for idx, dev_info in enumerate(devices):
            serial = dev_info.GetSerialNumber() if dev_info.GetSerialNumber() else "N/A"
            model = dev_info.GetModelName() if dev_info.GetModelName() else "N/A"
            print(f"Device {idx}: Serial Number={serial}, Model Name={model}")

        # Clear existing lists
        cameras.clear()
        converters.clear()
        capturing_flags.clear()
        saving_flags.clear()
        img_counts.clear()
        imgs.clear()
        start_flags.clear()
        camera_model_names.clear()

        if device_count == 0:
            print("No cameras available for connection.")
            camera_connected = False
            return False

        # Create an InstantCamera for each device
        for i, dev_info in enumerate(devices):
            cam = pylon.InstantCamera(tl_factory.CreateDevice(dev_info))
            cam.Open()

            model_name = cam.GetDeviceInfo().GetModelName()
            
            # If the model name contains "a2A2448", set pixel format to Mono8 with manual bit depth configuration (for a2A2448 series)
            if "a2A2448" in model_name:
                try:
                    # Set pixel format to Mono8
                    cam.PixelFormat.SetValue("Mono8")
                    # Set sensor bit depth mode to Manual
[... 156 lines cut ...]
def start_capture(i):
    if i >= len(capturing_flags):
        print(f"[start_capture] Camera index {i} is out of range.")
        return

    if not capturing_flags[i]:
        capturing_flags[i] = True
        t = Thread(target=capture_images, args=(i,))
        t.daemon = True  # The thread will also exit when the program terminates
        t.start()
        print(f"[Camera {i}] Capture started")
```

```
def stop_capture(i):
    if i < len(capturing_flags):
        capturing_flags[i] = False
        print(f"[Camera {i}] Capture stopped")

```

Could you please advise if there's anything else I should consider or adjust in my system setup or code to resolve the "pixel format not supported" error? Any further guidance on optimizing bandwidth or camera configuration in this multi-camera network setup would be greatly appreciated.
I have also applied the firewall settings and implemented the quick fix (setting PixelFormat to Mono8, BslSensorBitDepthMode to Manual, and BslSensorBitDepth to Bpp12), but I am still continuously experiencing grab fail errors.

Thank you for your help!

thiesmoeller, 2025-02-18:
You operate two 5G Cameras and one 1G camera on a single 1G network card.
-> You have to manage the camera bandwidth to get this working!

question about the setup:
And what is the expected framerate?
Do all cameras run in sync using HW trigger?

shimens, 2025-02-19:
In our future setup, we plan to use a high-performance PC and a 5G network card. Our ultimate goal is to achieve over 200fps with the a2A640-240gmSWIR camera, over 60fps with the a2A2840-67g5mUV camera, and 100–150fps or more with the a2A2448-105g5cBAS camera using ROI settings (2448×300).

Additionally, at present, all cameras operate in free-run mode when the capture start button is pressed, and the hardware trigger is used to send signals to an external STM32 board (for lighting control). 
(we have not yet tested the functionality of sending signals via the hardware trigger to the STM32.)

Would it be likely that the errors such as "pixel format currently not supported" or "grab failed" would disappear in a high-performance 5G network environment?

shimens, 2025-02-19:
I recently connected three cameras to a PC equipped with a 5G network card. As soon as the a2A2840-67g5mUV and a2A2448-105g5cBAS cameras are connected simultaneously, the following errors occur:

`Error 2025-02-19 14:29:12.673 Basler a2A2840-67g5mUV (40539233) Image acquisition on "Basler a2A2840-67g5mUV (40539233)" failed! Error: ""
Error 2025-02-19 14:29:12.672 Basler a2A2840-67g5mUV (40539233) Image acquisition on "Basler a2A2840-67g5mUV (40539233)" failed! Error: ""
Error 2025-02-19 14:29:12.672 Basler a2A2840-67g5mUV (40539233) Image acquisition on "Basler a2A2840-67g5mUV (40539233)" failed! Error: "The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e., the network adapter, switch, or Ethernet cable. Buffer underruns can also cause image loss. To fix this, use the pylonGigEConfigurator tool to optimize your setup and use more buffers for grabbing in your application to prevent buffer underruns."
Error 2025-02-19 14:29:12.641 Basler a2A2448-105g5c (40560945) Image acquisition on "Basler a2A2448-105g5c (40560945)" failed! Error: "The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e., the network adapter, switch, or Ethernet cable. Buffer underruns can also cause image loss. To fix this, use the pylonGigEConfigurator tool to optimize your setup and use more buffers for grabbing in your application to prevent buffer underruns."
`

Compared to using a 1G network, the errors are more severe on the 5G network card, and now the UV camera is also encountering issues. Could you please advise on what might be causing this and how to resolve it?

Thank you very much for your assistance.

thiesmoeller, 2025-02-19:
The fps will only lower the average bandwidth your system has to handle.

You have to lower the peak bandwidth of your cameras.
The speed in which they output one single frame.

And yes if the over temperature protection kicks in the camera can't operate normally.

Please monitor the device temperatures from your code and take care that you respect the thermal specs of each of the devices.

shimens, 2025-02-19:
Alright, I understand for now. Thank you for your response, and I'll reach out if any issues come up in the future.