basler/pypylon #822: Basler Gige cameras encounter unrecoverable exceptions after hours of continuous acquisition

borders, 2025-02-18:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I have a collection of 14 Basler Ace gige cameras (acA4112-8gm). I use a `Pylon::CBaslerUniversalInstantCameraArray` to interface with the cameras. I configure all of the cameras to be software triggered via action commands. After initializing all the cameras, I call `StartGrabbing()`, and then periodically I trigger the cameras using `IssueActionCommand()`. And I acquire images from the cameras using `Pylon::CBaslerUniversalInstantCameraArray::RetrieveResult()`. This all works very well, except that after several hours of continous operation (acquiring from the cameras every ~2 seconds) I often encounter an exception, which I believe is thrown by `RetrieveResult()`. The specific exception seems to vary. Often, it is:

```
Failed to read memory at 0x40464, 0x4 bytes. The control channel is already disconnected. (0xE1000016)
```

But I occasionally see these too:

```
Basler camera exception: Control channel not open.
Basler camera exception: Failed to read memory at 0x40464, 0x4 bytes. Timeout, no message received. (0xE101800B)

```

And, maybe not surprisingly, once I encounter that exception, I no longer receive images from the camera that threw the exception.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

* 14 Basler Ace gige cameras (acA4112-8gm)
* Zyxel XGS1930-28HP switch
* 10gbps PCIE NIC
* Linux workstation (x86_64, 64GB RAM)

### Camera(s) used

* 14 Basler Ace gige cameras (acA4112-8gm)
  * All have version: 107411-16 V2.1-0.

### Runtime information:

```shell
pylon 7.4.0.14900
```

borders, 2025-02-18:
I've seen similar-looking issues from the past (like #281), with comments saying that it is likely due to network configuration error. But I have configured the network settings as described here: https://docs.baslerweb.com/network-configuration-(gige-cameras)

borders, 2025-02-18:
I should have mentioned that I am using the Pylon c++ library - not the python wrapper of pylon. I assume it's all the same under the hood, but please let me know if I should take this issue somewhere else.

thiesmoeller, 2025-02-18:
Hi @borders,

Yes there is no difference of the lower layers between pypylon and native c++ pylon SDK.

What you see is the communication link closed between application and camera due to lost heartbeat packets or timeouts in active control communication.

Some first hints 
In your multicamera scénario you have quite a burst datarate into the host. 

So make sure you have enabled flow control on the switch ports and on the nic.

As you only need .5Hz you could also throttle the output datarate of the cameras via GevSCPD

Normal support channel would be for this your local support team to guide you in optimizing your system

closed 2025-02-20 by borders