basler/pypylon #825: How to get AutoBrightnessStatus chunkdata?

dk-teknologisk-bgd, 2025-03-10:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am trying to get the AutoBrightnessStatus information in the chunk data. It works well with ExposureTime and Gain, but not AutoBrightnessStatus.
I initialize like this:
``` for chunk in ["ExposureTime", "Gain", "AutoBrightnessStatus"]:
                self.camera.ChunkSelector.Value = chunk
                self.camera.ChunkEnable.Value = True ```

But after an image Grab, I am not able to read out the AutoBrightnessStatus:
``` meta = {"timestamp": timestamp,
                "exposureTime": grabResult.ChunkExposureTime.Value,
                "gain": grabResult.ChunkGain.Value,
                "autoBrightnessStatus": grabResult.ChunkAutoBrightnessStatus.Value} ```

I get the error:
``` AttributeError: no attribute 'ChunkAutoBrightnessStatus' in GrabResult ```

Does anyone know how to read out the ChunkAutoBrightnessStatus value? 

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Jetson orin NX 8 GB

### Camera(s) used

daA3840-45uc

### Runtime information:

```shell
python: 3.10.16 (main, Dec  4 2024, 08:53:37) [GCC 9.4.0]
platform: linux/aarch64/5.10.192-tegra
pypylon: 4.1.0 / 9.0.3.215
```

HighImp, 2025-03-10:
Hi, AutoBrightnessStatus is not a standard in the "Standard Feature Naming Convention" and basler specific, so its ""BslChunkAutoBrightnessStatus"

Please try: 
```python
meta = {
    "timestamp": timestamp,
    "exposureTime": grabResult.ChunkExposureTime.Value,
    "gain": grabResult.ChunkGain.Value,
    "autoBrightnessStatus": grabResult.BslChunkAutoBrightnessStatus.Value
}
```

dk-teknologisk-bgd, 2025-03-10:
@HighImp that worked! thanks for your support, im closing this issue

closed 2025-03-10 by dk-teknologisk-bgd