basler/pypylon #830: USB 3.0 camera image transport  not working properly

junhui, 2025-03-21:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Model: a2A5320-23umPRO , connect to PC USB 3.0, i use python to do line 1 trigger the shot. it give me error with *image transport not working properly*

## Exception
it throw exeption at `grab_result = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)`

the issue is my camera is using USB 3.0, not GigE network, anything wrong?

> Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped. : TimeoutException thrown (file 'instantcameraimpl.h', line 1058)

## Code

nothing special, no own logic yet. just follow the example code:

```python

import os
import time
from pypylon import pylon
 
# Define the folder to save images
save_folder = 'captured_images'
os.makedirs(save_folder, exist_ok=True)
 
try:
    # Initialize the camera
    tlFactory = pylon.TlFactory.GetInstance()
    devices = tlFactory.EnumerateDevices()
   
    if len(devices) == 0:
        print("No camera found. Please connect a camera.")
        exit(1)
   
    # Create and open the camera
    camera = pylon.InstantCamera(tlFactory.CreateDevice(devices[0]))
    camera.Open()
    print(f"Connected to camera: {camera.GetDeviceInfo().GetModelName()}")
   
    # Configure trigger safely
    try:
        # Method that works with all versions of pypylon
        try:
            # TriggerSelector
            camera.TriggerSelector.SetValue("FrameStart")
            print("Set TriggerSelector to FrameStart")
           
            # TriggerMode
            camera.TriggerMode.SetValue("On")
            print("Set TriggerMode to On")
           
            # TriggerSource
            camera.TriggerSource.SetValue("Line1")
            print("Set TriggerSource to Line1")
           
            # TriggerActivation
            camera.TriggerActivation.SetValue("RisingEdge")
            print("Set TriggerActivation to RisingEdge")
        except Exception as e:
            print(f"Error setting trigger parameters: {e}")
            print("Some trigger parameters might not be available on this camera model.")
    except Exception as e:
        print(f"Error configuring trigger: {e}")
   
    # Print camera configuration
    print("\nCamera configuration:")
    print(f"- Model: {camera.GetDeviceInfo().GetModelName()}")
    print(f"- Serial: {camera.GetDeviceInfo().GetSerialNumber()}")
   
    # Start grabbing images
    camera.StartGrabbing(pylon.GrabStrategy_OneByOne)
    print("Camera is waiting for triggers on Line1...")
   
    # Counter for naming images
    image_counter = 0
   
    while camera.IsGrabbing():
        try:
            # Wait for trigger with timeout
            grab_result = None
            try:
                grab_result = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

PC or embedded system model/type:
- CPU architecture: X86_64
- Operating System: windows 11
- RAM: 16GB

Interfaces used to connect the cameras:
- USB 3.0 
- cable: https://docs.baslerweb.com/basler-power-io-cable-m8-6p-open-p

Sensor:
- Omron E3Z-D62 

Wire connection: 
- Basler camera connect to PC USB 3.0 
- M8 6P white wire indicated line one connect to sensor black wire for signal 
- M8 6P blue wire connect with sensor blue with negative power 
- sensor brown connect with positive power 

### Camera(s) used

Basler a2A5320-23umPRO (40605386)
p=vu3pc_imx542m_pro/s=r/v=2.5.1/i=9421.6/h=e7c7a35

### Runtime information:

```shell
python: 3.10.6 (tags/v3.10.6:9c7b4bd, Aug  1 2022, 21:53:49) [MSC v.1932 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 4.1.0 / 9.0.3.215
```

HighImp, 2025-03-21:
Hi, thank you very much for the detailed description of your wiring!

I believe you need to add an additional pull-up resistor to get your application running—your code looks good.
You can monitor your I/O state by printing out the [Line Status](https://docs.baslerweb.com/line-status) to see if the line toggles at all.

But let me explain why I think you need a pull-up:

If this is your sensor wiring of the E3Z-D62 (NPN-Model): 
[image]

and this is the camera:
[image]

The sensor switches against GND, and there is no way for Line1 to reach the high state. 
 
I guess it would have worked for Line1 out of the box with the E3Z-D82 (PNP-Model) 
[image]

Anyway, to see if I am right: Could you please connect the sensor to Line2 or Line3 (GPIO lines) for a short test and adapt your code accordingly? Don’t forget to use the pink wire instead of the blue one for GND. Due to the built-in pull-up of the GPIO, this should work.

In case galvanic decoupling is required, please connect a resistor (e.g., 470 Ohm for 12V / 1k for 24V) between the black and brown cables of the sensor, to pull the state to "high" in idle mode.

To trigger an image correctly, you can either change your trigger activation to "FallingEdge" or activate the inverter of Line1. 
This is also necessary if you use Line2 or Line3.

Hope this helps!

junhui, 2025-03-21:
thank you so much. it works now by following your instructions

only one problem is each time got triggered, it capture too much photos. any better idea can limit the photo captured?

HighImp, 2025-03-22:
The Camera only takes one image per falling edge, so may your signal bounces. I did not read the datasheet of your sensor, is there a way to Limit the Output speed?
 That would be the best solution. To ignore too fast pulses inside the camera, you can reduce the fps by using the acquisition frame rate ( https://docs.baslerweb.com/acquisition-frame-rate) and set it a little higher than your expected trigger rate. 
E.g. if you expect one valid sensor Impulse per second, set it to 2. The camera will ignore every pulse 500ms after the last valid pulse.  
But i recommend the sensor side solution if possible. 

Just to be sure that it is not emv: Did you use Line1 or the gpios?

junhui, 2025-03-24:
Thank you. 
- I switched to Line 2 right now. 
- My sensor is E3Z-R61, it doesn't give option to limit the output speed, looks like I have to limit the acquisition-frame-rate or change a new sensor. 

Another challenge is as long as the object is moving (very slow moving), it have some level of blur. 
I changed ExposureTime, and Gain a lot of time, no matter in my code or through pylon viewer, I couldn't get a clear image if moving 

[image]

here is my code example for the camera settings:
```
            # TriggerSelector
            camera.TriggerSelector.SetValue("FrameStart")
            print("Set TriggerSelector to FrameStart")
            
            # TriggerMode
            camera.TriggerMode.SetValue("On")
            print("Set TriggerMode to On")
            
            # TriggerSource
            camera.TriggerSource.SetValue("Line2")
            print("Set TriggerSource to Line2")
            
            # TriggerActivation
            camera.TriggerActivation.SetValue("FallingEdge")
            print("Set TriggerActivation to FallingEdge")

            camera.ExposureTime.SetValue(8000)
            print("Set ExposureTime to 8000")
            # Set Gain to 16
            camera.Gain.SetValue(10)
            print("Set Gain to 10")
            # Enable frame rate control
            camera.AcquisitionFrameRateEnable.SetValue(True)
            print("Enabled frame rate control")
            
            # Set frame rate to 2 FPS
            camera.AcquisitionFrameRate.SetValue(2.0)
            print("Set AcquisitionFrameRate to 2 FPS")
```

HighImp, 2025-03-24:
Hi, and thank you for your trust! I believe your issue is heading in a direction where you should contact your local support or sales team, as they can understand and assist with your entire system much better.

It seems you're on the right track by reducing the exposure time, especially since the image is only blurry when the target is moving. However, 8000 µs is still quite a long exposure time. You might want to try increasing the lighting and raising the gain as much as possible within acceptable limits. If that doesn’t help, you might need to consider using a flash.

Regarding your "multi-trigger": Did the FPS limitation solve the issue? The [Input Filter](https://docs.baslerweb.com/input-filter) might also be a good solution, as it can help suppress false impulses at the I/O level.

As I mentioned, to ensure you receive the best support for your complete setup, please contact your local support team.