basler/pypylon #833: Consider Adding Type Stubs (.pyi) for _pylon.pyd and _genicam.pyd to Support IDE Autocomplete and Type Checking

qaz10102030, 2025-05-07:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I would like to kindly request support for type stubs (`.pyi` files) for the `_pylon.pyd` and `_genicam.pyd` modules included in the `pypylon` package.

**Issue**:
These modules are built as compiled C++ extensions and currently do not expose any static type information. When using an editor like Visual Studio Code with `"python.analysis.typeCheckingMode": "strict"`, the lack of `.pyi` files leads to missing type inference, autocompletion, and frequent type-checking errors — even for standard usage patterns such as accessing camera parameters via `InstantCamera`.

**Steps to Reproduce**:

1. Install `pypylon` in a Python project using Visual Studio Code.
2. Enable `"python.analysis.typeCheckingMode": "strict"` in `settings.json`.
3. Attempt to write typical usage code such as:

   ```python
   from pypylon import pylon
   camera = pylon.InstantCamera()
   camera.Open()
   camera.Width.SetValue(1024)
   ```
4. Observe that VS Code does not recognize members like `Width` or `SetValue()`, and shows type-checking warnings due to lack of type information.

**Request**:
If possible, please consider providing `.pyi` stub files or integrating inline type annotations for these modules. This would greatly enhance developer experience, improve compatibility with static analysis tools, and reduce the likelihood of runtime errors — especially in production environments where strong typing is enforced.

Thank you very much for considering this request.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

CPU: X86_64 / OS: Win10 / RAM: 32GB

### Camera(s) used

Basler acA2040-90umNIR (25195259)
106555-30;U;acA2040_90u;V4.1-1;0

### Runtime information:

```shell
python: 3.11.9 (tags/v3.11.9:de54cf5, Apr  2 2024, 10:12:12) [MSC v.1938 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 4.1.0 / 9.0.3.215
```

bjoernrennfanz, 2025-05-07:
Hi, thank you for your request.

The reason why you have no autocompletion, that these are dynamic parameters, which depend on the camera being used. You could check the Static API in the pylon C++ API or consult the Basler Product Documentation on (https://docs.baslerweb.com/knowledge/how-to-find-parameter-names-of-camera-features), the names are the same in Python.

thiesmoeller, 2025-05-07:
On https://github.com/thiesmoeller/pypylon-stubs I created some stubs a while ago.

Yes it is in our backlog to ship it with Pypylon 

The easiest way to work with pypylon is in Jupyter notebook. There you'll get dynamic auto completion after opening the camera

qaz10102030, 2025-05-08:
Thank you both for your responses!

I understand that it's possible to refer to the documentation for parameter names and usage, but I still believe there’s strong value in having officially maintained .pyi stub files for the pypylon API — especially to improve IDE support, type checking, and development experience with tools.

While community stubs are helpful, official support would provide greater confidence in compatibility and long-term usability.
I hope this can be considered as part of the core distribution in the future.

adwellj, 2025-05-08:
Hi, I'm also in this situation, where I would like improved type checking and autocompletion in my IDE.

@thiesmoeller, for me, I would be fine with the first option you presented (use InstantCamera with all features).

Thanks for your consideration and engagement!

@qaz10102030, thanks for opening this issue and providing such concise information!

qaz10102030, 2025-05-14:
Same as @adwellj, I also chose option 1.
If a feature isn’t supported by the device, perhaps a mechanism (e.g. an UnsupportedFeatureException or similar) could help catch it at runtime.

If Option 2 is ever considered, maybe it could be done at a series level rather than per exact model. For example, AreaScan models (Basler ace, dart, boost…), SWIR, TOF, etc., often share similar feature sets with incremental differences. Grouping wrappers by series might help maintain clarity without too much fragmentation and possibly even highlight missing or redundant features across product lines.
Not sure if this is feasible, if it's an unrealistic idea, feel free to ignore it :)