basler/pypylon #834: Camera not found by discovery

justwhiteecode, 2025-05-09:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi everyone,

I currently have a Basler GigE camera setted up. It is connected to a switch and the computers I want to access to the camera with is also connect to it (so they are in the same subnet). I am facing up some issues in the discovery: the script does not find anything. I mean, the ping works correctly and find the camera and also the pylon viewer detect it and is able to stream the video.

I had already setted all the necessary (I think): switch and computer's network interfaces are set with 9000 mtu and following the instructions in the readme. 

I have also provided to install pylon as recommended. Thinking it might be the problem, I started with version 8.1 and then downgraded on 7.5. and 7.3 but nothing solved the problem.

I do not know what to do more. I would be glad if you would help me.
Thank you

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

The machine runs ubuntu 24 and is x86 based.
# System Details Report
---

## Report details
- **Date generated:**                              2025-05-09 16:40:28

## Hardware Information:
- **Hardware Model:**                              Advantech UNO-148-B
- **Memory:**                                      32.0 GiB
- **Processor:**                                   11th Gen Intel® Core? i5-1145G7E × 8
- **Graphics:**                                    Software Rendering
- **Disk Capacity:**                               512.1 GB

## Software Information:
- **Firmware Version:**                            5.19
- **OS Name:**                                     Ubuntu 24.04.2 LTS
- **OS Build:**                                    (null)
- **OS Type:**                                     64-bit
- **GNOME Version:**                               46
- **Windowing System:**                            X11
- **Kernel Version:**                              Linux 6.11.0-25-generic

The switch is a cisco ie4000

### Camera(s) used

It does not print anything since it cannot find anything

### Runtime information:

```shell
python: 3.12.3 (main, Feb  4 2025, 14:48:35) [GCC 13.3.0]
platform: linux/x86_64/6.11.0-25-generic
pypylon: 4.1.0 / 9.0.3.215
```

SMA2016a, 2025-05-09:
would you make a tcp dump ?

justwhiteecode, 2025-05-09:
> would you make a tcp dump ?
@SMA2016a 

Here in the middle I tried to execute the script with pypylon:
`
19:15:25.261655 IP 192.168.37.150.319 > 224.0.1.129.319: PTPv2, v1 compat : no, msg type : sync msg, length : 44, domain : 0, reserved1 : 0, Flags [two step], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 15092, control : 0 (Sync), log message interval : 0, originTimeStamp : 108119 seconds, 552950288 nanoseconds
19:15:26.252005 IP 192.168.37.150.319 > 224.0.1.129.319: PTPv2, v1 compat : no, msg type : sync msg, length : 44, domain : 0, reserved1 : 0, Flags [two step], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 15093, control : 0 (Sync), log message interval : 0, originTimeStamp : 108120 seconds, 542973392 nanoseconds
19:15:27.073724 IP 192.168.37.150.320 > 224.0.1.129.320: PTPv2, v1 compat : no, msg type : announce msg, length : 64, domain : 0, reserved1 : 0, Flags [none], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 40316, control : 5 (Other), log message interval : 1, originTimeStamp : 0 seconds 0 nanoseconds, origin cur utc :37, rsvd : 0, gm priority_1 : 128, gm clock class : 248, gm clock accuracy : 254, gm clock variance : 65535, gm priority_2 : 128, gm clock id : 0x3053fffe36ee9b, steps removed : 0, time source : 0xa0
19:15:27.241778 IP 192.168.37.150.319 > 224.0.1.129.319: PTPv2, v1 compat : no, msg type : sync msg, length : 44, domain : 0, reserved1 : 0, Flags [two step], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 15094, control : 0 (Sync), log message interval : 0, originTimeStamp : 108121 seconds, 532954544 nanoseconds
19:15:27.343565 ARP, Request who-has 192.168.37.101 tell 192.168.37.150, length 46
19:15:27.343584 ARP, Reply 192.168.37.101 is-at cc:82:7f:33:32:18, length 28
19:15:27.344127 ARP, Request who-has 192.168.37.50 tell 192.168.37.150, length 46
19:15:27.344129 ARP, Reply 192.168.37.50 is-at cc:82:7f:33:32:18, length 28
19:15:27.631283 ARP, Request who-has 192.168.37.101 tell 192.168.37.150, length 46
19:15:27.631297 ARP, Reply 192.168.37.101 is-at cc:82:7f:33:32:18, length 28
19:15:27.631831 ARP, Request who-has 192.168.37.50 tell 192.168.37.150, length 46
19:15:27.631836 ARP, Reply 192.168.37.50 is-at cc:82:7f:33:32:18, length 28
19:15:28.237287 IP 192.168.37.150.319 > 224.0.1.129.319: PTPv2, v1 compat : no, msg type : sync msg, length : 44, domain : 0, reserved1 : 0, Flags [two step], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 15095, control : 0 (Sync), log message interval : 0, originTimeStamp : 108122 seconds, 522977560 nanoseconds
19:15:29.051972 IP 192.168.37.150.320 > 224.0.1.129.320: PTPv2, v1 compat : no, msg type : announce msg, length : 64, domain : 0, reserved1 : 0, Flags [none], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 40317, control : 5 (Other), log message interval : 1, originTimeStamp : 0 seconds 0 nanoseconds, origin cur utc :37, rsvd : 0, gm priority_1 : 128, gm clock class : 248, gm clock accuracy : 254, gm clock variance : 65535, gm priority_2 : 128, gm clock id : 0x3053fffe36ee9b, steps removed : 0, time source : 0xa0
19:15:29.221824 IP 192.168.37.150.319 > 224.0.1.129.319: PTPv2, v1 compat : no, msg type : sync msg, length : 44, domain : 0, reserved1 : 0, Flags [two step], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 15096, control : 0 (Sync), log message interval : 0, originTimeStamp : 108123 seconds, 512955752 nanoseconds
19:15:30.212011 IP 192.168.37.150.319 > 224.0.1.129.319: PTPv2, v1 compat : no, msg type : sync msg, length : 44, domain : 0, reserved1 : 0, Flags [two step], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 15097, control : 0 (Sync), log message interval : 0, originTimeStamp : 108124 seconds, 502982024 nanoseconds
19:15:31.035414 IP 192.168.37.150.320 > 224.0.1.129.320: PTPv2, v1 compat : no, msg type : announce msg, length : 64, domain : 0, reserved1 : 0, Flags [none], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 40318, control : 5 (Other), log message interval : 1, originTimeStamp : 0 seconds 0 nanoseconds, origin cur utc :37, rsvd : 0, gm priority_1 : 128, gm clock class : 248, gm clock accuracy : 254, gm clock variance : 65535, gm priority_2 : 128, gm clock id : 0x3053fffe36ee9b, steps removed : 0, time source : 0xa0
19:15:31.201867 IP 192.168.37.150.319 > 224.0.1.129.319: PTPv2, v1 compat : no, msg type : sync msg, length : 44, domain : 0, reserved1 : 0, Flags [two step], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 15098, control : 0 (Sync), log message interval : 0, originTimeStamp : 108125 seconds, 492956824 nanoseconds
19:15:32.196457 IP 192.168.37.150.319 > 224.0.1.129.319: PTPv2, v1 compat : no, msg type : sync msg, length : 44, domain : 0, reserved1 : 0, Flags [two step], NS correction : 0, sub NS correction : 0, reserved2 : 0, clock identity : 0x3053fffe36ee9b, port id : 1, seq id : 15099, control : 0 (Sync), log message interval : 0, [... cut ...]
19:20:01.664700 IP 192.168.37.150.49152 > 192.168.37.101.33892: BCM-LI-SHIM: direction unknown, pkt-type unknown, pkt-subtype unknown, li-id 14
19:20:01.665054 IP 192.168.37.150.49152 > 192.168.37.101.33892: BCM-LI-SHIM: direction unknown, pkt-type unknown, pkt-subtype unknown, li-id 14
19:20:01.665055 IP 192.168.37.150.49152 > 192.168.37.101.33892: BCM-LI-SHIM: direction unknown, pkt-type unknown, pkt-subtype unknown, li-id 14
19:20:01.665056 IP 192.168.37.150.49152 > 192.168.37.101.33892: BCM-LI-SHIM: direction unknown, pkt-type unknown, pkt-subtype unknown, li-id 14
`