basler/pypylon #835: Preferred process for image capture and parameter adjustment in seperated processes

mirza298, 2025-05-09:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello, I am developing an application where my Basler camera needs to grab images simultaneously while allowing parameter adjustments. I've considered using separate Python processes—one for image grabbing and another for parameter adjustment. However, I've encountered a problem: I can't pickle and share a camera object between processes. As I am new to multiprocessing with Python, I would appreciate seeing a similar example of how to perform this task with Basler cameras.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

CPU: X86_64
Ubuntu 22.04
RAM: 64Gb

### Camera(s) used

 a2A3536-31ucPRO_2 (40600739)

p=vu3pc_imx676c_pro/s=r/v=1.0.0/i=9722.9/h=e6b5eaf 

### Runtime information:

```shell
python: 3.10.16 (main, Dec 11 2024, 16:24:50) [GCC 11.2.0]

platform: linux/x86_64/6.8.0-59-generic

pypylon: 4.1.0 / 9.0.3.215
```

thiesmoeller, 2025-05-09:
What you try to do is typically not needed.

See these sample that describes the background loop capture mode https://github.com/basler/pypylon-samples/blob/main/notebooks/grabstrategies.ipynb

Here the capturing happens in real system threads outside of the python GIL only when you captured a new image control returns into your python code via the described OnImageGrabbed handler

mirza298, 2025-05-10:
Oh, thank you for the info. I was overcomplicating my code; the ImageHandler class is a really handy way of grabbing images.