basler/pypylon #836: BUG: Crash when calling CreateFirstDevice() with Microsoft Store Python on Windows 11

ahrensburger, 2025-05-15:
### Describe the issue:

I’m encountering a crash when using pypylon with a freshly installed Python 3.13 from the Microsoft Store on Windows 11. The issue does not occur when using Python from [python.org](http://python.org/).

- The TlFactory instance is created successfully.
- Calling CreateFirstDevice() causes an unhandled exception and crashes the Python interpreter entirely.
- This only happens when using the Microsoft Store version of Python.
- Using Python from [python.org](http://python.org/) works perfectly with the same code and environment.

Environment:

- Windows 11 (fully updated)
- Python 3.13 from Microsoft Store
- pypylon (latest version via pip)
- Basler pylon SDK (latest Windows version)

Suspected Cause:

This may be related to the sandboxed nature of Microsoft Store Python installations. These apps are restricted from accessing certain parts of the file system or external DLLs, which might interfere with pypylon. Logging or device discovery mechanisms may be attempting access to protected areas.

### Reproduce the code example:

```python
from pypylon import pylon

tl_factory = pylon.TlFactory.GetInstance()
print(f'{tl_factory}')

# This causes an unhandled exception and crashes the Python interpreter
dev = tl_factory.CreateFirstDevice()
print(f'{dev}')

camera = pylon.InstantCamera(dev)
print(f'{camera}')
```

### Error message:

```shell

```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

ace2 USB.

### Runtime information:

```shell
Given in the ticket.
```

hnebelung, 2026-09-16:
> (venv313) c:\Users\HNebelun\projects\repo_pypylon>python issue_836\crash.py
> <pypylon.pylon.TlFactory; proxy of <Swig Object of type 'Pylon::CTlFactory *' at 0x000002D0F14BA8B0> >
> <pypylon.pylon.PylonDevice; proxy of <Swig Object of type 'Pylon::IPylonDevice *' at 0x000002D0F14B9590> >
> <pypylon.pylon.InstantCamera; proxy of <Swig Object of type 'Pylon::CInstantCamera *' at 0x000002D0F14BB5A0> >
> Using device: a2A1920-51gcBAS

hnebelung, 2026-09-16:
Can't reproduces this bug with pypylon 26.8

closed 2026-09-16 by hnebelung