basler/pypylon #837: How to close camera correctly

Ting-Jia-Hu, 2025-05-19:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,

Sorry for my poor English.

Following is my code, I want to take pictures with checkerboard to do camera calibration, but when I press 'q' to stop grabbing image, it will show the error:

```
terminate called without an active exception
Aborted (core dumped)
```
this is my terminal

[image]

Is there something I miss to close or anything to do?
Please assist me in solving this problem.
Thank you and sorry for my poor English again.

```python
import cv2
import numpy as np
import glob
from pypylon import pylon
import matplotlib.pyplot

i=0

try:
    # connect the camera and print the model
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    print('using device',camera.GetDeviceInfo().GetModelName())

    #enter which camera do you want to calibrate
    camera_number = input('Which camera do you want to take pictutre :')

    # start grabbing the image
    camera.Open()
    camera.PixelFormat.SetValue("BGR8")
    camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

    converter = pylon.ImageFormatConverter()
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

    while camera.IsGrabbing():  

        grabResult = camera.RetrieveResult(5000,pylon.TimeoutHandling_ThrowException)

        try:
            if grabResult.GrabSucceeded():
                image = converter.Convert(grabResult)
                img = image.GetArray()
                # img = cv2.cvtColor(grabResult.Array,cv2.COLOR_BAYER_BG2BGR_EA)     #gray to colorful
                img0 = cv2.resize(img,(960,600))
                cv2.imshow('grabbing',img0)
        
                # if press 'q', then close the camera, and if press 's', then save a picture
                presskey = cv2.waitKey(1)
                if presskey == ord('s'):
                    i=i+1                                                                       # i is a variable,
                    cv2.imwrite('/home/csp41506/gan_ni_niang/camera_calibration/camera%s/checkerboard/%s.png'%(camera_number,i),img)  # then picture's name
                                                                                                                                    # is i=1,2,3...
                if presskey == ord('q'):    
                    print('stop')
                    cv2.waitKey(10)
                    break  
        finally:
            grabResult.Release()

except Exception as e:
    print("Error occurred",str(e))

finally:
    # stop grabbing the image
    cv2.destroyWindow('grabbing')
    if camera.IsGrabbing():
        camera.StopGrabbing()
        print("camera stop grabbing")
    if camera.IsOpen():
        camera.Close()
        print("camera is closed")
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Embedded system : NVIDIA Jetson Orin NX 16G
Interface : USB cable

### Camera(s) used

a2A1920-160ucBAS

### Runtime information:

```shell
pylon Viewer 7.4.0.14100
```

thiesmoeller, 2025-05-21:
Hi @Ting-Jia-Hu, your english is perfect!

What is your pypylon version? As we have fixed some potential shutdown issues in latest releases.

As your code is a quite standard request here is some code, that is doing the same.

important changes to your code is:

DestroyAllWindows to really close everything 

And use

with camera.RetrieveResult as ...

pattern to automatically delete the buffer

and use pick ( pip install pick ) for nice commandline interface

```python
import cv2
import numpy as np
from pypylon import pylon
import os
from typing import Optional, Tuple, List, TypeAlias
from pick import pick  # for commandline selection

# Define a named type for camera selection to improve readability and avoid repetition
CameraSelection: TypeAlias = Optional[Tuple[int, pylon.DeviceInfo]]

def select_camera_from_menu() -> CameraSelection:
    """
    Display an interactive menu to select a camera.
    Returns the selected camera index and its device info, or None if no camera is selected.
    """
    tlFactory = pylon.TlFactory.GetInstance()
    devices: List[pylon.DeviceInfo] = tlFactory.EnumerateDevices()

    if not devices:
        print("No camera devices found.")
        return None

    menu_options: List = []
    for i, device in enumerate(devices):
        model = device.GetModelName()
        serial = device.GetSerialNumber()
        menu_options.append(f"Camera {i+1}: {model} (S/N: {serial})")

    menu_options.append("Exit")

    try:
        selected_option, menu_index = pick(
            menu_options,
            title="Select a camera:",
            indicator="▶",  # Custom indicator for pick
        )
    except KeyboardInterrupt:
        # Handle Ctrl+C during menu selection gracefully
        print("\nMenu selection cancelled.")
        return None

    # Check if user selected Exit
    if menu_index == len(menu_options) - 1:
        return None

    return menu_index, devices[menu_index]

def capture_images(camera_selection: Tuple[int, pylon.DeviceInfo]) -> None:
    """
    Function to capture and save images from a Basler camera.
    Uses proper context managers for resource handling.

    Args:
        camera_selection: A tuple containing the selected camera's index and its DeviceInfo object.
    """
    camera_index, device_info = camera_selection

    image_counter = 0

    model_name = device_info.GetModelName().replace(" ", "_")
    serial_number = device_info.GetSerialNumber()
    folder_name = f"{model_name}_{serial_number}"

    save_path = f"./camera_images/{folder_name}/checkerboard/"

    os.makedirs(save_path, exist_ok=True)

    camera: Optional[pylon.InstantCamera] = None
    try:
        tlFactory = pylon.TlFactory.GetInstance()
        camera = pylon.InstantCamera(tlFactory.CreateDevice(device_info))

        camera.Open()

        print(f"Using device: {camera.GetDeviceInfo().GetModelName()}")
        print(f"Serial number: {camera.GetDeviceInfo().GetSerialNumber()}")
        print(f"Saving images to: {save_path}")
        print("Press 's' to save an image, 'q' to quit")

        camera.PixelFormat.SetValue("BGR8")

        converter = pylon.ImageFormatConverter()
        converter.OutputPixelFormat = pylon.PixelType_BGR8packed
        converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

        camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

        while camera.IsGrabbing():
            with camera.RetrieveResult(
                5000, pylon.TimeoutHandling_ThrowException
            ) as grab_result:
                if grab_result.GrabSucceeded():
                    image = converter.Convert(grab_result)
                    img = image.GetArray()

                    img_display = cv2.resize(img, (960, 600))
                    cv2.imshow("Image Capture", img_display)

                    key = cv2.waitKey(1)

                    if key == ord("s"):
                        image_counter += 1
                        image_path = f"{save_path}{image_counter}.png"
                        cv2.imwrite(image_path, img)
                        print(f"Saved image {image_counter} to {image_path}")

                    elif key == ord("q"):
                        print("Stopping capture")
                        break

    except Exception as e:
        print(f"Error occurred: {str(e)}")

    finally:
        if camera and camera.IsGrabbing():
            camera.StopGrabbing()
        if camera:
            camera.Close()
        cv2.destroyAllWindows()
        print("Capture session ended")

if __name__ == "__main__":
    camera_selection_result: CameraSelection = select_camera_from_menu()

    if camera_selection_result:
        capture_images(camera_selection_result)
    else:
        print("No camera selected. Exiting program.")
```

thiesmoeller, 2025-05-26:
This is a really old pypylon version.

Please do a

`pip install pypylon --upgrade`

Ting-Jia-Hu, 2025-05-28:
After I upgrade my pypylon version, it was  work correctly,
Thank you.

closed 2025-05-28 by Ting-Jia-Hu