basler/pypylon #841: Can't stream from multiple cameras on different network using multicast

isabellahuang, 2025-05-29:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi all, I was able to get a stream through Pypylon of a camera on a different network than my own via multicast, following [this thread](https://github.com/basler/pypylon/issues/572#issuecomment-1473089225) 

My computer is on the 192.168.1.x network, and my cameras are on the 10.20.0.x network

However, I now want to stream from two cameras, and I am finding that this code only streams successfully from the last camera in my list (i.e. index = 1). Printing debugging statements show the following order of actions:

- grabResult comes from camera 0 (fails)
- grabResult comes from camera 1 (succeeds)
- grabResult comes from camera 1 (succeeds)
- grabResult comes from camera 1 (succeeds)
- ..... and it continues on using just camera 1

How can I get a stream from both cameras, instead of just one? 

```
from pypylon import pylon
from PIL import Image 

# Identifying information of the cameras

# Seems to only stream from the last ip address in this list
ip_addresses = ['10.20.0.51', '10.20.0.52']
serial_numbers = ["40638508", "40638510"]

# Create a camera array
cameras = pylon.InstantCameraArray(len(ip_addresses)) 

# Create and attach all Pylon Devices.
for i, cam in enumerate(cameras):
    tlf = pylon.TlFactory.GetInstance()
    tl = tlf.CreateTl('BaslerGigE')
    cam_info = tl.CreateDeviceInfo()

    cam_info.SetIpAddress(ip_addresses[i])
    # cam_info.SetSerialNumber(serial_numbers[i]) # This doesn't work

    cam.Attach(pylon.TlFactory.GetInstance().CreateFirstDevice(cam_info))
    print("Using device ", cam.GetDeviceInfo().GetModelName())

cameras.Open()
for cam in cameras:
    cam.StreamGrabber.TransmissionType.Value ="Multicast"
    cam.StreamGrabber.DestinationAddr.Value = "192.168.1.5"

numberOfImagesToGrab = 100
cameras.StartGrabbing(numberOfImagesToGrab)

while cameras.IsGrabbing():
    grabResult = cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    cameraContextValue = grabResult.GetCameraContext()

    if grabResult.GrabSucceeded():
        cameraContextValue = grabResult.GetCameraContext()

        # Print the index and the model name of the camera.
        print("Camera ", cameraContextValue, ": ", cameras[cameraContextValue].GetDeviceInfo().GetIpAddress())
```

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup used

PC or embedded system model/type:
- CPU architecture X86_64
- Operating System Ubuntu 22.04.2 LTS
- RAM 251 Gb

Interfaces used to connect the cameras:
- Type in type/model/name of interface cards
- Switches or hubs used
- Cable types/lengths

### Camera(s) used

Two cameras

Basler dmA2448-23gc (40638508)
p=mg2b_imx547c/s=r/v=1.0.0/i=9803.31/h=6c58501

and 

Basler dmA2448-23gc (40638510)
p=mg2b_imx547c/s=r/v=1.0.0/i=9803.31/h=6c58501

### Runtime information:

```shell
pylon info: 10.2.1.471 64-Bit
----------------------------------------
pylon Setup info: 8.1.0.16743 
----------------------------------------
pylon Application info: 4.0.2.16804 64-Bit
----------------------------------------
pylon Viewer info: 10.0.2.16804 64-Bit
----------------------------------------
Host system infos:

CPU architecture: x86_64
Kernel type: linux
Kernel version: 5.19.0-41-generic
OS type: ubuntu
OS version: 22.04
Render Backend: OpenGL
----------------------------------------
Screen(s):

Screen 0 scale factor: 1
Screen 0 resolution: 3840x1600
----------------------------------------
Window(s):

"pylon Viewer 64-Bit" on screen 0
----------------------------------------
Host network infos:

Host network adapter eno1:
    Name: eno1
    MAC address: 3C:EC:EF:D1:8A:CA
    IP address: 192.168.1.5
    Subnet mask: 255.255.0.0

Host network adapter eth0:
    Name: eth0
    MAC address: 3C:EC:EF:D1:89:F8

Host network adapter ens4f0:
    Name: ens4f0
    MAC address: B4:96:91:EE:24:08
    IP address: 192.167.1.114
    Subnet mask: 255.255.255.0

Host network adapter enxb03af2b6059f:
    Name: enxb03af2b6059f
    MAC address: B0:3A:F2:B6:05:9F
    IP address: 169.254.3.1
    Subnet mask: 255.255.255.0

Host network adapter ens4f1:
    Name: ens4f1
    MAC address: B4:96:91:EE:24:09
    IP address: 10.110.21.96
    Subnet mask: 255.255.252.0

Host network adapter docker0:
    Name: docker0
    MAC address: 02:42:C4:F5:88:29
    IP address: 172.17.0.1
    Subnet mask: 255.255.0.0

Python information:
python: 3.12.9 | packaged by Anaconda, Inc. | (main, Feb  6 2025, 18:56:27) [GCC 11.2.0]
platform: linux/x86_64/5.19.0-41-generic
pypylon: 4.1.0 / 9.0.3.215
```

isabellahuang, 2025-06-03:
I looked closer at the images stream and it appears that there is some data mixing happening after the first pair of images. For example, here is the sequence of camera images I'm getting, where I am getting `grabResult = cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)` in a loop

- grabResult.GetCameraContext() from cam 0, data from cam 0 , grabResult.ImageNumber = 1
- grabResult.GetCameraContext() from cam 1, data from cam 1, grabResult.ImageNumber = 1
- grabResult.GetCameraContext() from cam 1, data from cam 0, grabResult.ImageNumber = 2
- grabResult.GetCameraContext() from cam 1, data from cam 0, grabResult.ImageNumber = 3
- grabResult.GetCameraContext() from cam 1, data from cam 0, grabResult.ImageNumber = 4, ...
- etc. 

Only the first two grab results seem correct, and from that point on, it's like it mixes the two camera results together.

isabellahuang, 2025-06-06:
Hi @thiesmoeller,

Thank you so much for your response. I'm running this new code snippet now based on your suggestions
```

from pypylon import pylon
from PIL import Image

# Identifying information of the cameras

# Seems to only stream from the last ip address in this list
ip_addresses = ['10.20.0.51', '10.20.0.52']

# Create a camera array
cameras = pylon.InstantCameraArray(len(ip_addresses))

# Create and attach all Pylon Devices.
for i, cam in enumerate(cameras):

    tlf = pylon.TlFactory.GetInstance()
    tl = tlf.CreateTl('BaslerGigE')
    cam_info = tl.CreateDeviceInfo()

    cam_info.SetIpAddress(ip_addresses[i])
    cam.Attach(pylon.TlFactory.GetInstance().CreateFirstDevice(cam_info))

    print("Using device ", cam.GetDeviceInfo().GetModelName())
    print(cam.GetDeviceInfo().GetFriendlyName())

cameras.Open()
for i, cam in enumerate(cameras):
    cam.StreamGrabber.TransmissionType.Value = "Multicast"
    cam.StreamGrabber.DestinationAddr.Value = f"224.1.1.{i+1}"

    cam.Width.Value = 100
    cam.Height.Value = 100
    
    cam.SetCameraContext(i)

numberOfImagesToGrab = 100
cameras.StartGrabbing(numberOfImagesToGrab, pylon.GrabStrategy_OneByOne)

while cameras.IsGrabbing():
    try:
        with cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
            if grabResult.GrabSucceeded():
                cameraContextValue = grabResult.GetCameraContext()
                image_number = grabResult.ImageNumber
                
                print(f"Camera {cameraContextValue}: IP {cameras[cameraContextValue].GetDeviceInfo().GetIpAddress()}, Image #{image_number}")
                # Process image here
            
            else:
                print(f"Grab failed: {grabResult.GetErrorDescription()}")
                
    except pylon.TimeoutException:
        print("Timeout occurred")
        break
```

However, I get the following error:

```
Traceback (most recent call last):
  File "/home/isabellahuang/pypylon/samples/isabella_multicam_cleaned.py", line 42, in <module>
    with cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/isabellahuang/miniconda3/lib/python3.12/site-packages/pypylon/pylon.py", line 4683, in RetrieveResult
    return _pylon.InstantCameraArray_RetrieveResult(self, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
_genicam.RuntimeException: There is already a thread waiting for a result. : RuntimeException thrown (file 'InstantCameraImpl.h', line 1090)
```

Would you know how I can solve this issue?

thiesmoeller, 2025-06-06:
Had the wrong start grabbing all

```python
from pypylon import pylon
from PIL import Image

# IP addresses of the cameras
ip_addresses = ['10.20.0.51', '10.20.0.52']

# Create a camera array
cameras = pylon.InstantCameraArray(len(ip_addresses))

# Create and attach devices
for i, cam in enumerate(cameras):
    tlf = pylon.TlFactory.GetInstance()
    tl = tlf.CreateTl('BaslerGigE')
    cam_info = tl.CreateDeviceInfo()
    cam_info.SetIpAddress(ip_addresses[i])
    cam.Attach(pylon.TlFactory.GetInstance().CreateFirstDevice(cam_info))
    print("Using device:", cam.GetDeviceInfo().GetModelName())
    print("Friendly name:", cam.GetDeviceInfo().GetFriendlyName())

# Open cameras and configure
cameras.Open()
for i, cam in enumerate(cameras):
    cam.StreamGrabber.TransmissionType.Value = "Multicast"
    cam.StreamGrabber.DestinationAddr.Value = f"224.1.1.{i+1}"
    cam.Width.Value = 100
    cam.Height.Value = 100
    cam.SetCameraContext(i)

# Set target number of images
target_images = 100
image_counts = {i: 0 for i in range(len(ip_addresses))}

# Start grabbing indefinitely
cameras.StartGrabbing(pylon.GrabStrategy_OneByOne)

try:
    while cameras.IsGrabbing():
        with cameras.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException) as grabResult:
            if grabResult.GrabSucceeded():
                cam_index = grabResult.GetCameraContext()
                image_counts[cam_index] += 1
                print(f"Camera {cam_index} (IP {cameras[cam_index].GetDeviceInfo().GetIpAddress()}): "
                      f"Image #{image_counts[cam_index]}")

                # Stop grabbing for individual camera if target reached
                if all(count >= target_images for count in image_counts.values()):
                    print("Reached target image count for all cameras.")
                    cameras.StopGrabbing()
                    break

            else:
                print(f"Grab failed: {grabResult.GetErrorDescription()}")
except pylon.TimeoutException:
    print("Timeout occurred")

# Clean up
cameras.Close()

```

isabellahuang, 2025-06-09:
Hi @thiesmoeller, I should clarify. I have two cameras on a 10.20.1.x network connected to
computer A. The cameras are perfectly accessible through computer A due to
being on the same subnet.

Computer A and computer B are also connected via a 192.168.1.x network. My
goal is to stream from both cameras on computer B.

The cameras are on a different subnet from computer B, so they were not
directly accessible from computer B. I saw from previous GitHub issues that
multicasting was the suggested method to stream to a different subnet (by
setting DestinationAddr to the IP of computer B).

I was indeed able to get the stream of a single camera from computer B
using multicast, but I am not able to successfully stream from multiple
cameras. I appear to get some strange mixing of the packet data from the
two cameras.

thiesmoeller, 2025-06-16:
Have you tried first to get the cameras working just with pylon viewer and the "Add Remote GigE Camera" menu entry.

Like the idea  I raised before, the multicast trick was for ace1 you have a ace2 camera with more advanced networking support.

If this works you can do the same from pypylon side

isabellahuang, 2025-06-18:
Hi @thiesmoeller , I am able to add it through "Add Remote GigE Camera". It shows up as a device but I cannot stream through pylon Viewer. No images show up. Are there any steps I need to take besides entering its IP address on the 10.20.0.x network when prompted to from "Add Remote GigE Camera"?

isabellahuang, 2025-06-18:
@thiesmoeller What transmission type should I be using? Also, do I need to start up two pylon Viewer instances for a controlling and a monitoring application?

SMA2016a, 2025-07-02:
Please do not use the GigE transport layer (TL) directly to create a DeviceInfo object.

'''
tl = tlf.CreateTl('BaslerGigE')  # This will not stream properly!
tlf = pylon.TlFactory.GetInstance()
cam_info = tl.CreateDeviceInfo()   
'''

'''
cam_info = pylon.DeviceInfo()
cam_info.SetIpAddress(ip_addresses[i])
cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateDevice(cam_info))
'''

A few troubleshooting questions:

    Do you have a router that routes IP packets between the host and the camera (and vice versa)?

    Can you ping the camera successfully?

 If yes, try using "Add Remote GigE Camera" in your setup instead.

isabellahuang, 2025-07-07:
Thanks @thiesmoeller and @SMA2016a 

I am configuring a laptop as a router that routes packets between the host and the camera. 
I am also able to ping the camera successfully. I can successfully "Add Remote GigE Camera" on PylonViewer, but the stream is empty. 

I haven't been able to get past this step at all.

SMA2016a, 2025-07-08:
the next step would be to open the camera, set the destination IP in the streaming parameter to IP of the host system.  To set that you need to 

'''
camera.GetStreamGrabberParams().TransmissionType.SetValue(TransmissionType_Multicast);
camera.GetStreamGrabberParams().DestinationAddr.SetValue("192.168.3.11");
GenICam::gcstring s = camera.GetStreamGrabberParams().DestinationAddr.GetValue();

'''

thiesmoeller, 2025-07-08:
In your first post you enabled multicast.
This recommendation is not touching this control. And leave it to unicast

bjoernrennfanz, 2025-07-16:
@isabellahuang I suspect it's the destPort, you have to set a separate port for each camera.

```
# Create a camera array
cameras = pylon.InstantCameraArray(len(ip_addresses)) 

...

cameras.Open()
for i, cam in enumerate(cameras):
    cam.StreamGrabber.TransmissionType.Value = "Multicast"
    cam.StreamGrabber.DestinationAddr.Value = f"224.1.1.{i+1}"
    cam.StreamGrabber.DestinationPort.Value = f"{i+5555}"

...

```