basler/pypylon #844: Software triggerd multi camera capture - Pixel format currently not supported

yairhs, 2025-06-18:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello,

I running a multicamera setup. all cameras are Basler ace acA5472-5gm.
Cameras are connected to a Basler GigE Interface Card, 4 Port PoE.
Cameras are software triggered.
I occusianly get "Pixel format currently not supported" errors during grabbing.
This can happen after ~1000 captures. But can also happen multiple times in a raw.

I have a different PC using a Basler 5GigE Interface Card, 4 Port PoE+ with the same code where I dont see this error.

This is asnippet of my code:
```
  try:
      grab = cam.RetrieveResult(4000, pylon.TimeoutHandling_ThrowException)
      if not grab.GrabSucceeded:
          err = grab.GetErrorDescription()
          self.logger.error("Grab failed on %s: %s", cid, err)
          grab.Release()
          raise RuntimeError(err)
    
      # Extract image and metadata
      img = grab.GetArray()
      w, h = grab.GetWidth(), grab.GetHeight()
      ox, oy = grab.GetOffsetX(), grab.GetOffsetY()
      grab.Release()
```
This is the error:
```
  File "pypylon\pylon.py", line 167, in func_wrapper
  File "pypylon\pylon.py", line 2250, in GetArray
  File "pypylon\pylon.py", line 167, in func_wrapper
  File "pypylon\pylon.py", line 2232, in GetImageFormat
ValueError: Pixel format currently not supported
```
Is there a way to minimize this type of errors?
Is this a hardware issue?

Thank you

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

PC
Win 11
X64

### Camera(s) used

Basler ace acA5472-5gm

### Runtime information:

```shell
python: 3.12.8 (main, Jan 14 2025, 22:49:36) [MSC v.1942 64 bit (AMD64)]
platform: win32/AMD64/11
pypylon: 4.1.0 / 9.0.3.215
```

yairhs, 2025-06-25:
@SMA2016a Thank you for the suggestion.
I optimized the network parameters using the pylon GigE configurator and it ssems to help. I am still running continuously to verify that the error doesn't happen again.

What happens if I disconnect the cameras? Are the parameters set by configurator saved on the camera? or this needs to be done again after disconnecting from power

SMA2016a, 2025-07-20:
Try to set the packet size to 8000 and check if this works otherwise try to ping with packet size 8000