basler/pypylon #845: Camera grab images in a wrong size using pypylon in Python

NicoleCu, 2025-07-21:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello everybody,
I am from Germany, and I am setting data acquisition via Python with my new a2A2448-210cm camera. The default resolution is 2448 x 2048, but the camera grabs images only with height 1024, while I need a full default resolution. I tried to change a ROI by
``` 
camera.Height.Value = 2448 
camera.OffsetX.Value = 0
camera.OffsetY.Value = 0 
``` 
but it gives out the following problem: `` LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16815)``

Please recommend me a way to resolve this issue.  

 

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Microsoft Windows 11 Enterprise

### Camera(s) used

a2A2448-210cm

### Runtime information:

```shell
python: 3.12.4 (tags/v3.12.4:8e8a4ba, Jun  6 2024, 19:30:16) [MSC v.1940 64 bit (AMD64)]
platform: win32/AMD64/11
pypylon: 4.2.0 / 10.2.1.471
```

NicoleCu, 2025-07-22:
Hi! I tried this code, but the situation has not changed unfortunately. The output is 
`` Image shape: (1024, 2464)``

thiesmoeller, 2025-07-22:
Check if the camera boots from factory settings:

```python
from pypylon import pylon

cam = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
cam.Open()

# Set the user set to default as first step
cam.UserSetSelector.Value = "Default"
cam.UserSetLoad.Execute()

cam.OffsetX.Value = 0
cam.OffsetY.Value = 0
cam.Width.Value = cam.Width.Max
cam.Height.Value = cam.Height.Max

grab = cam.GrabOne(5000) # Timeout in ms

if grab.GrabSucceeded():
	img = grab.Array # NumPy-Array
	print("Image shape:", img.shape) # (height, width)
else:
	print("Error:", grab.ErrorCode, grab.ErrorDescription)

grab.Release()
cam.Close()
```

NicoleCu, 2025-07-23:
Thank you very much for your help! It finally works.