basler/pypylon #848: Synchronized StartAcquisition of multiple cameras using ActionCommands

hirschenberger, 2025-08-06:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hey, I have a 8 `Racer 2 r2L4096-84g5m` 5GigE Linescan cameras that I want to start at the same time.

I want to use ActionCommands to do this.

I realized that there's no possibility to select `AcquisitionStart` as `TriggerSelector` for `Action1`.

I can only select `FrameStart` as selector. This works but unfortunately only grabs one frame.

How can I start synchronized continuous acquisition with action commands?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Debian 13

### Camera(s) used

8 x r2L4096-84g5m

### Runtime information:

```shell
Pylon 8.0.2 64bit Debian Linux
```

SMA2016a, 2025-08-07:
The Action Command is designed for scenarios where the application sends an action command for each frame.
More information: [Action Commands – Basler Documentation](https://docs.baslerweb.com/action-commands)

However, you might be looking for a Periodic Trigger instead.
See: [Periodic Signal – Basler Documentation](https://docs.baslerweb.com/periodic-signal)

hirschenberger, 2025-08-07:
I have 8 cameras looking at an endless material belt. Left and right side, dark- and brightfield, upper and lower side.

All camera's coordinate systems are mapped to a common coordinate system where error locations are found.

The cameras are attached to a common pulsetrain as line-trigger.

For the coordinate-system transformations which are predetermined and later applied online, it is necessary that all cameras start acquisition deterministically.

I try to avoid using a hardware IO-Signal when it would be possible to do it in software.

If I remember correctly, on older Basler Linescan cameras it was possible to select `AcquisitionStart` as `TriggerSelector` for actions, right?

thiesmoeller, 2025-08-07:
Yep. That's true.

Have to check with the racer team for a workaround.

Will be e.g. FrameActive Trigger and keep it active from a timer or a counter as a side effect of your action command.

thiesmoeller, 2025-08-11:
please try:

FrameBurstTriggerStart -> Source -> Action1
FrameBurstTriggerStart -> TriggerMode -> On
FrameBurstTriggerEnd -> Source -> SoftwareSignal1
FrameBurstTriggerEnd -> TriggerMode-> On

Of course you could also use a second Action signal to stop frame generation...

Disclaimer, i haven't validated this but to give you a fast indication for a possible fix ... you get the workaround idea hot out of the press.