basler/pypylon #850: EnumerateDevices() sometimes terminates the application with core dumped

jokubas11, 2025-08-12:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Never had this kind of issue before, but calling EnumerateDevices() sometimes (not always) following message:

main.py
```
camera = CameraHandler()
cams = camera.get_cameras_list()
```

basler_camera.py
```
class CameraHandler:
    def __init__(self) -> None:
        ...

    @staticmethod
    def get_cameras_list() -> list[tuple[pylon.CDeviceInfo, str]]:
        """
        Return the list of Basler cameras available.

        Returns:
            List[Tuple[pylon.CDeviceInfo, str]]: Returns a list of
                tuples containing two items:
                1. Camera descriptor, used to create camera object
                2. String name, useful if needed to list out all cameras
        """
        tf = pylon.TlFactory.GetInstance()
        lst = [(cam, cam.GetModelName()) for cam in tf.EnumerateDevices()]
        return lst
```

OUTPUT:

```
(.venv) jokubasz@PCNAME:~/projects/basler_handler$ python main.py 
[(<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x738570ba9260> >, 'acA2040-90umNIR')]
(.venv) jokubasz@PCNAME:~/projects/basler_handler$ python main.py 
[(<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x721573faea30> >, 'acA2040-90umNIR')]
(.venv) jokubasz@PCNAME:~/projects/basler_handler$ python main.py 
[(<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x792008dcb810> >, 'acA2040-90umNIR')]
(.venv) jokubasz@PCNAME:~/projects/basler_handler$ python main.py 
[(<pypylon.pylon.DeviceInfo; proxy of <Swig Object of type 'Pylon::CDeviceInfo *' at 0x712b73fae820> >, 'acA2040-90umNIR')]
(.venv) jokubasz@PCNAME:~/projects/basler_handler$ python main.py 
pure virtual method called
terminate called without an active exception
Aborted (core dumped)
```

error always happens on call to
```
tf.EnumerateDevices()
```

Camera not an issue, happens without camera connected to PC as well
If this error doesn't happen, cameras work well, no issue with any of the cameras.
Everything works well in pylon viewer

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

PC: Linux PCNAME 6.14.0-27-generic 27~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 22 17:38:49 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
pypylon version: 4.2.0

I recently upgraded my OS from Ubuntu 22 to Ubuntu 24.
Not sure if it is related, but never had the same issue on Ubuntu 22.
Also, not camera related, happens even with camera plugged off.
Also, I have not developed with Basler cameras for a long time, so I am not sure if version has anything to do with this either.

### Camera(s) used

Not relevant, happens without camera plugged in.

### Runtime information:

```shell
python: 3.12.3 (main, Jun 18 2025, 17:59:45) [GCC 13.3.0]
platform: linux/x86_64/6.14.0-27-generic
pypylon: 4.2.0 / 10.2.1.471
```

BrendanMitchellSisu, 2026-09-04:
> Have you tried out our new collection of samples in pypylon 26.06, for instance info_dumper.py? Does the problem persist?

Yes, it does.

BrendanMitchellSisu, 2026-09-04:
I've got new  information after some investigation. 

## Summary

`pylon.TlFactory.GetInstance().EnumerateDevices()` intermittently aborts the whole Python interpreter with:

```
pure virtual method called
terminate called without an active exception
Aborted (core dumped)
```
In my testing, this occured in 20-30% of calls to `EnumerateDevices()`. This reproduces with a GigE camera, emulated hardware, or with nothing connected at all.   

Root cause, confirmed via a post-mortem core dump analysis (details below): pylon's GenTL Consumer layer (`libpylon_TL_gtc.so`) probes every GenTL producer listed in `GENICAM_GENTL64_PATH` on every `EnumerateDevices()` call. On any machine where Basler's CoaXPress producer (`ProducerCXP.cti`) is discoverable, that probe-and-close cycle races against the producer's own log4cpp-based logging library during teardown (`GCCloseLib()`), and one thread ends up calling a virtual method on an object another thread has already destroyed.

This bug only manifests if the pylon SDK deb package is installed in addition to `pypylon`. This is because `GENICAM_GENTL64_PATH` is populated by the system-wide pylon SDK installer.

## Environment

| Component | Version |
|---|---|
| OS | Ubuntu 24.04.4 LTS, x86_64 |
| Python | 3.12.3 |
| pypylon (pip) | 26.6 |
| System `pylon` package  | `pylon 26.06.2-deb0 amd64` |

## Reproduction

Install the pylon SDK version 26.06.2 on Ubuntu 24.04.   
Create a python 3.12.3 virtual environment and install `pypylon` version 26.6.

Run the following several times, starting a fresh process each time (just running `EnumerateDevices()` in a loop isn't sufficient):

```python
# enumerate_devices_crash.py
from pypylon import pylon

tlf = pylon.TlFactory.GetInstance()
devices = tlf.EnumerateDevices()
```

**Note:** For some reason, running under `gdb` (or `strace`) seems to suppress the race on my machine. 

## Core Dump

I let the crash occur with Ubuntu's `apport` crash handler enabled, then loaded the resulting core file into gdb post-mortem. Two threads are relevant:

**The crashing thread**: `__cxa_pure_virtual` called from inside the CXP producer's bundled logging library:

```
#0  __pthread_kill_implementation (...)
...
#4  __GI_abort ()
#5  ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  std::terminate() from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  __cxa_pure_virtual () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  ?? () from /opt/pylon/lib/pylonCXP/lib64/libcommon-logging-log4cpp.so
#10 start_thread (...)
```

**Another thread** tears down the producer:

```
#4  ?? () from /opt/pylon/lib/pylonCXP/lib64/libcommon-logging-log4cpp.so
#5  ?? () from /opt/pylon/lib/pylonCXP/lib64/libcommon-logging-log4cpp.so
#6  ?? () from /opt/pylon/lib/pylonCXP/lib64/libcommon-logging-log4cpp.so
#7  ?? () from /opt/pylon/lib/pylonCXP/lib64/libcommon-logging-log4cpp.so
#8  ?? () from /opt/pylon/lib/pylonCXP/lib64/libcommon-logging-dispatcher.so
#9  ?? () from /opt/pylon/lib/pylonCXP/bin/ProducerCXP.cti
#10 ?? () from /opt/pylon/lib/pylonCXP/bin/ProducerCXP.cti
#11 ?? () from /opt/pylon/lib/pylonCXP/bin/ProducerCXP.cti
#12 GCCloseLib () from /opt/pylon/lib/pylonCXP/bin/ProducerCXP.cti
#13 ?? () from .../pypylon/libpylon_TL_gtc.so
#14 ?? () from .../pypylon/libpylon_TL_gtc.so
#15 ?? () from .../pypylon/libpylon_TL_gtc.so
#16 ?? () from .../pypylon/libpylon_TL_gtc.so
#17 std::__shared_count<...>::~__shared_count() from .../pypylon/libpylon_TL_gtc.so
#18 GetTLInfoList () from .../pypylon/libpylon_TL_gtc.so
#19 ?? () from .../pypylon/libpylonbase.so.12
#20 Pylon::CTlFactory::InternalEnumerateTls(Pylon::TlInfoList&) from .../pypylon/libpylonbase.so.12
#21 Pylon::CTlFactory::EnumerateTls(Pylon::TlInfoList&) from .../pypylon/libpylonbase.so.12
#22 Pylon::CTlFactory::EnumerateDevices(Pylon::DeviceInfoList&, Pylon::DeviceInfoList const&, bool) from .../pypylon/libpylonbase.so.12
#23 Pylon::CTlFactory::EnumerateDevices(Pylon::DeviceInfoList&, bool) from .../pypylon/libpylonbase.so.12
#24 _wrap_TlFactory_EnumerateDevices () from .../pypylon/_pylon.abi3.so
...
#40 Py_BytesMain ()
```

Reading this chain: `TlFactory::EnumerateDevices()` → `EnumerateTls()` → `InternalEnumerateTls()` → `GetTLInfoList()` in the generic GenTL Consumer layer (`libpylon_TL_gtc.so`) → this loads every producer listed in `GENICAM_GENTL64_PATH` to ask what it supports, including `ProducerCXP.cti` → `GCCloseLib()` tears that producer down again once queried → its shutdown cascades into its log4cpp-based logging library → another thread concurrently calls a virtual method on an already-destroyed object.

## Workaround

The variable `GENICAM_GENTL64_PATH` is set by `/etc/profile.d/basler-gentl-path.sh`:

```bash
export GENICAM_GENTL64_PATH=$GENICAM_GENTL64_PATH:$basler_gentl_lib_dir
```

which comes from Basler's standard system-wide Pylon Software Suite installer; therefore one workaround is to simply `apt remove pylon`. `pip install pypylon` provides its own copy of the entire pylon runtime inside the wheel (`site-packages/pypylon/*.so`) and needs nothing from `/opt/pylon`. However, note that the `pypylon` [readme](https://github.com/basler/pypylon) says installing the Pylon Software Suite is "strongly recommended".

For any application that doesn't need CoaXPress support, another workaround is stripping the CXP producer path out of `GENICAM_GENTL64_PATH` before the first pylon `TlFactory` call in the process. This eliminates the crash.

```python
import os

path = os.environ.get("GENICAM_GENTL64_PATH", "")
if path:
    kept = [p for p in path.split(os.pathsep) if "cxp" not in p.lower()]
    os.environ["GENICAM_GENTL64_PATH"] = os.pathsep.join(kept)
```

This must run before any pylon call in the process.