basler/pypylon #852: Reset Camera Timestamp in Python

kranok-dev, 2025-08-20:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,

I would like to know how to reset the camera's timestamp in Python? 

Can't seem to find the equivalent of **GevTimestampControlReset** to Python from the documentation.

Thank you,

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

Nvidia Jetson

### Camera(s) used

Camera Models: a2A5320-23ucBAS and acA2040-55uc

### Runtime information:

```shell
platform: linux/aarch64/5.15.136-tegra
pypylon: 4.2.0 / 10.2.1.471
```

kranok-dev, 2025-08-26:
@Abhilashsabhi, not yet :(

@SMA2016a, it's USB camera. Is it possible to reset the timestamp from a USB camera without being powered off? I made a rough estimate given the size of the variable and tick count, and the variable would overflow in about 11 days. Do you know if it just simply starts at 0 again?

thiesmoeller, 2025-08-26:
@kranok-dev .. you saw the docu link ?

And one other advice. Typically you track this in the application. E.g. by matching it to system clock with a running mean or even a proper fitting over the last e.g. minute.( Cam time vs precision system time directly after RetrieveResult)  This way you can assign precision system time to each frame received

thiesmoeller, 2025-08-26:
The Docs are dynamic. Please select your camera in the top menu

kranok-dev, 2025-08-26:
Yes, I have put one of the cameras I'm using but I'm not sure how to call such function (**TimestampReset**) with Python:

[image]

From the camera object, I only find these functions:
- ...
- TestPendingAck
- TimerControl
- TimerDelay
- TimerDuration
- TimerReset
- TimerSelector
- TimerStatus
- TimerTriggerActivation
- TimerTriggerArmDelay
- TimerTriggerSource
- TimestampLatch
- TimestampLatchValue
- TransportLayer
- TransportLayerControl
- TriggerActivation
- TriggerDelay
- TriggerEventTest
- TriggerMode
- TriggerSelector
- TriggerSoftware
- TriggerSource
- ...

kranok-dev, 2025-08-27:
@thiesmoeller understood, I appreciate the follow-up and clarification, thank you.

Abhilashsabhi, 2025-08-31:
Please review the attached code and let me know if any issues are there in the code.