basler/pypylon #853: Difference between trigger and capture time tends to increase as cameras are left open

kranok-dev, 2025-09-13:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,

Maybe this is not the right place to ask, and I know this is the repo for the Python implementation of Pylon. I already tried the implementation in Python but ran into many issues with substantial use of RAM, hence decided to implement my code in C++. 

**My question is why the difference between trigger time and capture time tends to increase over time?** Is something building up on the cameras that is increasing the latency for capturing images? If so, is there a way to clear it without closing and reopening the cameras?

I get the datetime when I do the triggering and obtain the clock tick count from the images that are grabbed (I also do corresponding adjustment from clock tick count  in nanoseconds to seconds and compare it to a baseline clock tick count when attaching the cameras). 

My basic code, based on trial and error from sample codes and documentation.
```
void initiateCameras(CInstantCameraArray &cameras, CTlFactory &tl_factory, DeviceInfoList_t &devices){
    for(size_t i = 0; i < cameras.GetSize(); ++i){
        cameras[i].Attach( tl_factory.CreateDevice(devices[i]));
        cameras[i].RegisterConfiguration(new CSoftwareTriggerConfiguration, RegistrationMode_ReplaceAll, Cleanup_Delete);
        cameras[i].Open();

        INodeMap& nodemap = cameras[i].GetNodeMap();

        CEnumParameter(nodemap, "PixelFormat").SetValue("BayerRG8");
        CBooleanParameter(nodemap, "AcquisitionFrameRateEnable").SetValue(true);
        CFloatParameter(nodemap, "AcquisitionFrameRate").SetValue(10.0);
        CFloatParameter(nodemap, "ExposureTime").SetValue(5500.0);
        CEnumParameter(nodemap, "GainSelector").SetValue("All");
        CFloatParameter(nodemap, "Gain").SetValue(6);
    }
}

void processImage(CInstantCameraArray &cameras, int cam_index, CImageFormatConverter &formatConverter){
    CGrabResultPtr ptrGrabResult_2;
    cv::Mat cv_image;
    CPylonImage pylonImage;

    cameras[cam_index].RetrieveResult(1000, ptrGrabResult_2, TimeoutHandling_ThrowException);

    if(ptrGrabResult_2->GrabSucceeded()){
        intptr_t cameraContextValue = ptrGrabResult_2->GetCameraContext();
        auto nanoseconds_passed = ptrGrabResult_2->GetTimeStamp()-initial_clock_ticks[cameraContextValue];
        formatConverter.Convert(pylonImage, ptrGrabResult_2);
        cv_image = cv::Mat(ptrGrabResult_2->GetHeight(), ptrGrabResult_2->GetWidth(), CV_8UC3, (uint8_t *)pylonImage.GetBuffer());
        cv::imwrite("test.jpg", cv_image, {cv::IMWRITE_JPEG_QUALITY, 95}); 
    }
    else{
        cout << "Error: " << std::hex << ptrGrabResult_2->GetErrorCode() << std::dec << " " << ptrGrabResult_2->GetErrorDescription() << endl;
    }
}

void getImages(CInstantCameraArray &cameras){
    CGrabResultPtr ptrGrabResult_1;

    CImageFormatConverter formatConverter;
    formatConverter.OutputPixelFormat = PixelType_BGR8packed;

    for(uint32_t i = 0; i < cameras.GetSize() && cameras.IsGrabbing(); ++i){
        if (cameras[i].WaitForFrameTriggerReady(1000, TimeoutHandling_ThrowException)){
            cameras[i].ExecuteSoftwareTrigger();
            std::cout << "Trigger Time for " << cameras[i].GetDeviceInfo().GetModelName() << " " << cameras[i].GetDeviceInfo().GetSerialNumber() << " - ";
            printTime();
        }
    }
    cout << endl;

    std::this_thread::sleep_for(std::chrono::milliseconds(100));
    for(uint32_t i = 0; i < cameras.GetSize() && cameras.IsGrabbing(); ++i){
        processImage(cameras, i, formatConverter);
    }
    
}

int main(int, char* []){
    int exitCode = 0;
    
    PylonInitialize();

    try{
        CTlFactory& tl_factory = CTlFactory::GetInstance();

        DeviceInfoList_t devices;
        if(tl_factory.EnumerateDevices(devices) == 0){
            throw RUNTIME_EXCEPTION("No cameras present.");
        }

        CInstantCameraArray cameras(min(devices.size(),max_cameras));
        initiateCameras(cameras, tl_factory, devices);

        
        cameras.StartGrabbing(GrabStrategy_LatestImageOnly); 

        while(1){
            getImages(cameras);
        }

        cameras.StopGrabbing();
        cameras.Close();
    }
    catch(const GenericException& e){
        cerr << "An exception occurred." << endl << e.GetDescription() << endl;
        exitCode = 1;
    }

    cerr << endl << "Press enter to exit." << endl;
    while(cin.get() != '\n');

    PylonTerminate();

    return exitCode;
}
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup used

1 x Jetson Orin Nano

### Camera(s) used

2 x acA2040-55uc cameras
4 x a2A5320-23ucBAS cameras

### Runtime information:

```shell
python: 3.8.10 (default, Mar 18 2025, 20:04:55) 
[GCC 9.4.0]

platform: linux/aarch64/5.10.120-tegra

pypylon: 1.8.0 / 6.2.0.18677
```

HighImp, 2025-09-15:
Hi,
Do you compare the camera clock with your host PC’s clock?
If so, please keep in mind that two clocks never run at exactly the same speed and will always drift apart over time.
There are several ways to check the internal clock.
Please refer to:

https://docs.baslerweb.com/timestamp
https://docs.baslerweb.com/data-chunks#timestamp-chunk

thiesmoeller, 2025-09-16:
Hi @kranok-dev, 

the camera and your PC each use their own oscillator to keep time. Assume a typical precision of the oszillator of 50ppm or 100ppm.

e.g. 100ppm would mean that the oszillator could be up to 8.64 seconds too fast or too slow per day.
This is a mix of systematic and temperature dependant error.

Your PC is typically connected to a time server ( NTP ) that will correct this down to ~1 - 100 ms in relation to atomic clock standards.

So what you observe is exactly what is expected.
The solution for USB is to continously align the camera timestamps with the time of your PC. 
For GigEVision this is already implemented in the PTP support of the camera.